Ralim/IronOS #1004: TS80p keeps restarting

Zeehel, 2021-06-27:
Hello,

I bought a TS80p on Amazon. I wanted to use the soldering iron together with a QC/PD powerbank. I have tried two different powerbanks. But the soldering iron keeps restarting and for some reason it changes the output voltage of the powerbank.It switches between 5V and 9V but never up to 12V. On powerbanks have up to 12V PD/QC output.
Is there something wrong with my new TS80p? I have tried the original and custom firmware. The problem stays the same. Any ideas?

Ralim, 2021-06-27:
Please provide details on your power bank, what port you are correction too and what type of cake you are using.

Remember PD requires a C-C cable.

If the iron is restarting, where does this occur in bootup? Is this restarting once you solder, or once you start heating, or once it does the menu C or while it shows the miniware logo?

What power adapters and model numbers have you tried?

Zeehel, 2021-06-27:
What do you mean by "cake"?

With PD I use a type C-C cable. With QC a normal A-C cable.
I got two powerbanks:
[image]
[image]

It restarts willingly. Mostly while scrolling in the menu. When it the TS80p shows 9V I can heat it and it doesn't turn off.
But why it never drains 12V from the powerbank?

I tried different power adapters. All can use QC and PD. With two the TS80p uses 9.2V and with the other two it uses 12V.

Hope its enough information

Ralim, 2021-06-28:
> 
> 
> What do you mean by "cake"?

Ah heck, typo.. 😁  meant "cable"

> 
> With PD I use a type C-C cable. With QC a normal A-C cable.
> I got two powerbanks:
> [image]
> [image]

So over PD both of these banks will always only negotiate 9V as you need to support 2.6A @ 12V for the iron to select 12V.

> 
> It restarts willingly. Mostly while scrolling in the menu. When it the TS80p shows 9V I can heat it and it doesn't turn off.
> But why it never drains 12V from the powerbank?

It will never select 12V as these power banks do not support enough power output. 
In the menu you can override QC to use 12V, (and use an A->C cable) But when you do this it will randomly trip off whenever it attempts to heat the tip, as this will attempt to draw 2.6A which the power supply will see as an over current and thus turn off its output.

This will be occuring in the menu as the iron will by default pull a short pulse of power to keep the power bank awake (and stop it moving to sleep).

> 
> I tried different power adapters. All can use QC and PD. With two the TS80p uses 9.2V and with the other two it uses 12V.

What it negotiates via PD will depend on what current the adapter advertises. Even so, if you are _just_ rated enough power you may still run into issues if its a cheaper supply as they often dont like the pulsed load of the iron. Its best to use a power supply rated at _least_ 35W if you would like to use 12V.

> 
> Hope its enough information
> 
> Can I somehow mess with the "power pulse" settings to get rid of the probelm? -edit:
> 
> Okay I tried some options with the power pulse settings, and now it seems to work with my powerbanks.
> And even use 12V with QC :)
> 
> But I haven't found any specific informations/recommendatoins on how to use the power pulse settings.
> My settings are now: power pulse: 0.7V, delay 2, duration 4. Any objections?

What works for you on your adapters is what works for you 😁 
Though if you have had instability with the pulses be wary that it may trip out when soldering occasionally. 
I generally advise to use 9V if you dont need the extra power, as thats within the rating of the power bank (and thus you are not pushing its lifespan).

Alternatively there is a menu option to limit the power down. Running 12V, but with the power limited pulled down can often work better as it will allow you better recovery of temperature, but the iron will keep the average power under the limit so the power bank is less stressed.

> 
> And please allow me two other questions:
> 1.With IronOS I can use the TS80P with 5V even for soldering?! Or does it affect it in a negative way, if I use it for example with 5V/2A ?

It will just be really underpowered, as its a 4.5 Ohm tip, so it can only draw a little over 1A. So you are only running at around 5-6W.  It wont damage the iron at all, just doesnt work well. But good in an emergency. I have used it for a small fix powered from a laptop in desperation.

> 
> 2.Where can I contribute to you? Like tip for a beer or something like that :)

On the [front page of this repo](https://github.com/Ralim/IronOS), on the far right are the links :) 
All money donated generally gets invested back into tools for the project or coffee consumed while working on it 🤣.
[image]

Zeehel, 2021-06-28:
Thank you for your time for this detailed answer, Ralim.

I got the gist of how the soldering iron is operating.

I wanted to buy a 90W Powerbank from RAVPower yesterday with a PD 12V/3A Output, but unfortunatly they don't ship to germany right now :/

I would really appreciate if you could explain the power pulse options in detail, maybe with examples ?!^^. What does every single option affect in general.
I would like to unterstand the operating principle of the power pulse so I can optimize the use of my powerbank. And what negative effect does it have when I shorten the time or rise the voltage.

Another question, can you or anyone confirm that the powerbank (both of them for me) make some "white noise" or some kind of "cheeping" when the soldering iron is heating? Is it harmful for the MPB?

Thank you!

Zeehel, 2021-07-04:
Hello? Would you mind giving me an answer?! :) @Ralim

Ralim, 2021-07-05:
@Zeehel 

> I would really appreciate if you could explain the power pulse options in detail, maybe with examples ?!^^. What does every single option affect in general.
> I would like to unterstand the operating [...]

I dont have time to draw up examples, but it is your classical square wave.
You are adjusting the three primary values that define a square wave:

* How often a pulse occurs (frequency)
* How tall is the pulse (power)
* How wide is the pulse (duty cycle)

This is exposed with all three parameters as powerbanks are all different.
The amount of power "wasted" into the tip is the integral, or the area inside the square of the square wave.
As such you want to minimise this, but still have the bank stay on.

Some banks need a certain "wattage" to be seen to stay on, some need a certain duration, some want a certain amount of power drawn. As such generally you will want to adjust each one in turn to find the best settings, or just find one that works and leave it at that.

> 
> Another question, can you or anyone confirm that the powerbank (both of them for me) make some "white noise" or some kind of "cheeping" when the soldering iron is heating? Is it harmful for the MPB?

MPB?

The power bank will create some noise during operation. Due to how the soldering iron works it tends to draw power in small pulses of power. I.e. it goes 0%, 100%,0% fairly fast. This can cause noise in some powerbanks that are not designed for this and dont have filtering for this. Generally its not an issue, but if its really loud it may be concerning, but sadly the only thing you can really do is use a different power bank. I've not had it kill any power banks, but I could see if you had an especially cost-cut one or one that you were pushing the rating on, this pulsing behaviour may lead to a slightly earlier death due to wear and tear. Its not going to cause any sudden failures anytime soon so long as you stay within the rating.

White noise is totally normal, some laptops also do this :) Just and artefact of some DC/DC design.

> 
> Thank you!

> Hello? Would you mind giving me an answer?! :) @Ralim

Please dont bump threads :P 

I don't always watch my public email every day :) Replies will come eventually

discip, 2022-02-12:
@Zeehel
Is this issue still present?
If not please close. 😊

thanks

discip, 2022-04-24:
@Ralim 
You answered all his questions, he managed to solve the initial issue and does not respond.
Case closed, right? 😊

closed 2022-04-24 by Zeehel