Ralim/IronOS #1007: Pinecil not enumerating on USB

xobs, 2021-06-30:
**Describe the bug**
My Pinecil has stopped enumerating over USB. When I connect it, `dmesg` shows errors such as:

```
[ 1648.610862] usb 1-1: USB disconnect, device number 73
[ 1650.874269] usb 1-1: new full-speed USB device number 74 using xhci-hcd
[ 1651.010296] usb 1-1: device no response, device descriptor read/64, error -71
[ 1651.250301] usb 1-1: device no response, device descriptor read/64, error -71
[ 1651.490266] usb 1-1: new full-speed USB device number 75 using xhci-hcd
[ 1651.622318] usb 1-1: device no response, device descriptor read/64, error -71
[ 1651.862298] usb 1-1: device no response, device descriptor read/64, error -71
[ 1651.974488] usb usb1-port1: attempt power cycle
```

**To Reproduce**
Steps to reproduce the behavior:
1. Plug in Pinecil

**Expected behavior**
The device should enumerate

**Details on your device:**
 - Device: Pinecil G 78
 - Release: v2.15.40087E6 (https://github.com/Ralim/IronOS/releases/download/v2.15/Pinecil.zip)
 - Power adapter being used: USB port with BC12

Note you can grab these from the debug menu.

**Additional context**
My Pinecil stopped enumerating, so I updated the firmware to v2.15 by holding down the `-` button as I plugged it in. I then ran `dfu-util -d 28e9:0189 -a 0 -D Pinecil_EN.bin -s 0x08000000`. The Pinecil enumerated correctly and installed the updated firmware. It is for this reason that I believe it's a software problem.

Additionally, the device fails under both Windows and Linux on two different machines with different USB-C cables.

Ralim, 2021-06-30:
This firmware does not enumerate on USB under normal firmware. USB is only used for DFU firmware updates.

xobs, 2021-06-30:
I see. I must have a fundamental misunderstanding of how USB PD and USB BC1.2 work -- I thought they required enumeration.

There's a comment mentioned in https://github.com/Ralim/IronOS/issues/28 -- I thought the situation might have changed, but I suppose not. Did the patch to hold D+ high ever get added?

Ralim, 2021-06-30:
PD 2+ is communicated on the CC pins, entirely separate from the data pins. As such there is no requirement for the USB enumeration. (We all silently ignore PD 1.0).

That is a rather old issue 😅, but yes it was implemented but that only applies for the ts100.

For TS80,TS80P, and Pinecil; these all have QC. For QC2 and 3, you have to negotiate using voltages on the data pins. If a normal PC sees these it does what your original post says, in that it thinks a device is connecting but not answering, as it interprets the negotiations of this as a presence pull-up.

This never causes harm, just a little confusing for the PC.

closed 2021-06-30 by xobs

xobs, 2021-06-30:
Apologies. I thought maybe it used BC1.2 for negotiation, or something similar.

uis246, 2022-05-02:
Even USB PD 2+ can use Vbus communication

Ralim, 2022-05-02:
Vbus communication is not supported by the Pinecil or Type-C chargers