Ralim/IronOS #1008: The file has properties that can't be copied to the new location

hazymat, 2021-07-02:
Appreciate this may not be an issue with IronOS! Posting here for help. But may also be useful update the firmware install notes for IronOS to account for this in case others experience this.

- I'm in Win10 (21H1)
- Download the release, extract TS100_UK.hex, right-click on the file > properties > unblock
- Hold button A whilst connecting the iron to my laptop
- Drag the file onto USB drive
- Immediately the usb device disconnects and reconnects, but Windows shows this error: "Are you sure you want to copy this file without its properties? The file TS100_UK.hex has properties that can't be copied to the new location"

However by the time you go to clip Yes / Skip / Cancel, the drive has already disconnected and reconnected and shows the file TS100_UK.ERR

I have retried, assuming it's a timing issue have tried to hit "yes" really quickly, to no avail.

When I click "skip", after the drive reconnects, there's a file copy progress bar, but it doesn't appear to work and I'm left with two TS100_UK.ERR files on the drive.

Any hints?

Ralim, 2021-07-04:
I personally have no wisdom on this sadly.
I would suggest trying a google around about how you can strip the extra tracking properties that windows puts on downloaded files via some web-browsers.
I _think_ this may help? https://winaero.com/modify-or-remove-file-property-details-in-windows-10/

AlvaroHermida, 2021-07-05:
This is a wrongful marketing practice from Micro$oft: since OneDrive is a pityful failure and they cannot beat Dropbox and the other services which offer a much better experience than Micro$oft's half-cooked ripoff, they punish Dropbox users by blocking their workflow with this ridiculous "Are you sure you want to copy this file without its properties?" everytime they access a file secured by Dropbox, Inc.

To workaround this, try to copy the files first to some out-of-Dropbox folder, ("C:\finger4them\", for example) and then drag the file from this new folder.

And yes, it could be a good idea to update the install notes, so please tell us if this solves the issue, so we can do it.

hazymat, 2021-07-05:
@AlvaroHermida That was exactly the problem. Thank you. I moved the .hex file to the root of C:\ and was able to copy it across.

closed 2022-02-15 by Ralim