Ralim/IronOS #1009: USB-C PD Causes forced resets

Bradley-Lin, 2021-07-09:
**Describe the bug**
Device repeatedly resets itself after showing the Miniware logo for a split second. On the first time, it manages to make it to the idle screen for a split moment as well but this doesn't show again in future repeated resets.

**This issue does not happen with the stock firmware provided by Miniware**. Miniware's TS80P firmware works perfectly fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect one end of USB-C to USB-C cable to TS80P (for e.g. using the cable that comes in the box)
2. Connect the other end to a USB-PD wall charger (for e.g. using the wall supply that comes in the box)

**Expected behavior**
Device not resetting.

**Details on your device:**
 - Device: TS80P by Miniware
 - Release: 2.15.1A0B542
 - Power adapter being used: Miniware provided unbranded adapter. However this issue can be seen in other adapters.

Note you can grab these from the debug menu.

**Additional context**
Powering from a standard USB-A 5V port works fine, and the device will heat itself up from the 5V port.

Ralim, 2021-07-14:
@Bradley-Lin 

So to debug this we need to know the actual make and model of the power adapter being used, to be able to make a guess at to what its doing internally. Miniware's PD stack has bugs and assumptions in it, I'm sure the PD stack for this firmware also has invalid assumptions.

Also note that the chargers that come in a _lot_ of the TS80P kits are whatever was cheapest at the markets that day, there is no real reason as to what model is sent, so its very much up to luck.

To debug this, it needs to be reproducible on a charger where we can capture the PD negotiation.
Do you have any PD debugging tools?
Do you have links to the chargers that have this issue?

Bradley-Lin, 2021-07-14:
Unfortunately I don't have any PD debugging tools. I will be getting my own FUSB302 chip in the near future so I could probably debug that instead which should be similar. I can list the chargers:
"TRAVEL CHARGER"
Model: QC09
Supports 5V,9V,12V,15V at 3A
Supports 18V at 2.5A and 20V at 2.25A

My HP laptop charger is:
Regulatory model: TPN-TA02
HP part number: L45962-007
Artesyn Embedded Technologies model number: 700-014719-0001
Manufactured by ASTEC ELECTRONICS (LUODING) CO., LTD
Outputs: 20V @ 3.25A, 15V @ 4.33A, 12V @ 5A, 9V @ 3A, 5V @ 3A

Neither of them work.

autagi, 2021-07-16:
Hi Ralim,
thank you for your job!
I have the same issue of Bradley-Lin.
My power adapter is the one carried out with my TS80P by Miniware with your latest firmware.
Also the charger is the same of Bradley-Lin.
It works fine with this one.
https://www.amazon.it/gp/product/B085YGJXZR/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

A trick I found is this: power on the solder without solder tip and then connect the tip while the solder is powered and in this way it work fine.

DjordjeMandic, 2021-07-17:
I can confirm same issue on my TS80P with Samsung 25W USB C PD charger and the one that came with TS80P. 

TC66C Just reboots at same time with TS80P every few seconds(intervals are not fixed, for example if im in settings menu reboot interval is longer).

Seems like issue with PD code, QC works fine...

Ralim, 2021-07-18:
The samsung charger may be running into the issue that is documented elsewhere in the issues around here, that it _absolutely_ does not like the usb data pin toggling that the TS80P bootloader does. And will shutoff its output anytime the USB lines are active. It doesnt _always shutoff_, but unsure why.

I am slowly working on cleaning up and getting some more formalised testing around the PD library that I'll be bringing into Iron-OS at some point in the future, but I do not have a timeline on this.

https://github.com/Ralim/usb-pd/ is the pd code slowly being cleaned up and refactored

DjordjeMandic, 2021-07-18:
@Ralim i still dont get it why is the usb pd soo buggy, its not your fault, i think that its because every charger is implemented a little bit differently

Bradley-Lin, 2021-07-21:
I have a slight feeling this issue might be due to the sudden draw of current that causes a sudden drop in the supply voltage.

DjordjeMandic, 2021-07-23:
@Bradley-Lin if you wish i will test it with a precharged capacitor bank to same voltage that will be connected before starting heating process.

But i guess firmware already has a soft start feature where it ramps up power slowly.

But i don't think this might be a problem because its still reseting in basic standby menu

Ralim, 2021-07-23:
Firmware will soft start a little. But due to the hardware nature where we can only turn the tip on or off, and not do nice ramping.

Make sure you have power pulses turned off, then if it resets in menu its probably PD. if it stops resetting then its your charger.

Be wary adding extra capacitance as this can cause issues with PD chargers. (There is a specified maximum capacitance)

discip, 2022-03-31:
@Bradley-Lin
Is this solved running the latest commit?
If so, please close this issue. 😊

thanks

Bradley-Lin, 2022-04-01:
I'm actually pretty busy these days. I will be using the soldering iron at the end of next week so I will try it then.

discip, 2022-04-24:
@Bradley-Lin 
Did you have a chance to test the iron?

Bradley-Lin, 2022-04-24:
Sorry I got extremely busy and still am currently. I will try when I can. Unfortunately it might be a while.

brainard52, 2022-05-06:
I've noticed that the device reboots in a similar manner when it's plugged into my 2020 M1 MacBook Pro (Unfortunately Apple's website doesn't describe the power capabilities of the TBT3 ports other than "charging"). It shows either the power information or the boot logo. for a frame before it flickers.

My TS80p came with an unbranded power supply labeled "TRAVEL CHARGER" and "Model: QC09". When I have the TS80p with 2.18 into this, it appears to behave as expected.

I also have a portable battery from Mophie. When plugged into this, it initially reports 5 volts. When I press the "heat up" button, it heats on 5 volts for a few seconds before flashing the display. It then says 9 volts and heats as expected.

chixxi, 2022-06-02:
This one is also happening when connecting to my Macbook Air M1 using the TS80P on Release v2.18 (USB-C to USB-C Cable). This confused me  abit since it happened right after flashing the firmware, but it works just fine when using other power supplies.

BTW, works fine when connected to the Apple Studio Display.

Let me know if I can do aynthing to help figure out the caus of the issue. An obviously, a HUGE THANK YOU for this project!

D-314, 2024-01-15:
The problem is still present.

Device: TS80P by Miniware
Release: 2.22E.C308FE8C (05-11-23)
Power adapter being used: Xiaomi 65W dual port GaN Charger (AD652GEU), problem only with type-c port.

The device boots up, requests the required voltage (the screen shows 12V), and resets after a second. The frequency of resets is every 2~2.3 seconds.
It works fine with the Apple charger, but on the PD 9V profile, since the charger does not have a 12V