Ralim/IronOS #1014: Pinecil QC problems (undervoltage) & updating problems

JanezGroselj, 2021-07-17:
**Describe the bug**

#1 My new Pinecil reports undervoltage from several USB power sources which work perfectly fine on my TS80 (QC3.0 USB charger that I received with my TS80, Xiaomi 20000mah powerbank, bench PSU with a QC USB port)

#2 Pinecil firmware updater reports "DFU-Util exited with error code 74!" error

#3 Powering the Pinecil via a 12V 5A power supply, the Pinecil only uses 18W and is much slower to heat up than the TS80 on QC.

**To Reproduce**
Self explanatory I think.

**Expected behavior**
Self explanatory I think.

**Details on your device:**
 - Pinecil
 - Release: stock firmware (order about a month ago)
 - Power adapter being used: see above

JanezGroselj, 2021-07-17:
Regarding #2 I've managed to update to the latest 2.15 firmware with the GD32 MCU Dfu Tool, but sadly it has not fixed the other issues.

Ralim, 2021-07-17:
1) QC only negotiates after PD times out, are you waiting long enough for this to occur

2) Pinecil firmware updater is not a part of this repository, please seek help with its repository as I don't have the time to debug it as well 😅

3) all of the irons in use in this code are resistive devices.
Max power draw is a function of voltage. At 12V you are stuck at 17W or so. The tip is around 8 ohms. You can use ohms law to calculate the max power at any voltage. This is why 20V is recommend.

JanezGroselj, 2021-07-17:
Hi, thank you for the fast reply and thorough explanation, I really appreciate it!

1.  I'm waiting for a minute, hopefully that's enough 🙃
I honestly thought that the Pinecil would have actual working QC support (out of the box with stock firmware) like the TS80 as it is sold as such and that is why I bought several of them as gifts and QC powerbanks to go with them...
(I have my TS80 for almost two years now and QC is perfect for me, even though I know PD is more powerful)
I don't know if you are the right person to address this issues to, if not I'm sorry and please point me in the right direction if one exists, or have I just bought the wrong iron and should have gone with the TS80(P)  🙂

2. Sorry, I didn't realize that when I was writing 🙂

3. Oh thank you for the explanation! Yes I'm testing it on 20V and it uses a lot more watts. I guess the law is the law. 🙃

JanezGroselj, 2021-07-18:
So I have tried yet another QC charger, this one being capable of outputting 33W at a range of voltages from 5 to 20V and this time the Pinecil doesn't report undervoltage, yet it only goes up to 9V and 8.8W.
Could there be a problem with the latest batch of Pinecils?

Ralim, 2021-07-18:
So for QC, one of the key things to know is that its a spec in so much as the name exists, but unless your charger has the Qualcomm certification, there is ZERO promise it will follow their spec. Even to this, this code uses a slightly more flexible routine than the QC spec because the first version (to the spec) would only work on around 50% of chargers. Basically every non-certified chip uses its own slightly different flavour and quirks on the implementation. Its not possible to be compatible to all of them at once.

Adding into this; there is the issue of PD compliance as well. 
If a charger is compliant to the PD spec, it will not have QC on the same port.
If a charger is compliant to the PD spec it should ignore the USB pins.
There are a _lot_ of chargers out there that dont meet this, and instead will abort all QC if they see anything occuring on the usb pins. Additionally, _some_ will reset their output's power (and thus the iron) if this occurs.

Because of this, the TS80P and the Pinecil both work on the idea of delaying the QC negotiation until _after_ PD has timed out (given up). However, for a small subset of chargers, this doesnt work. As they _timeout_ their QC after around 2-3 seconds.

So its not possible to support both of these groups without manual toggles the user would need to make on a _different_ supply before using them.

So; in addition, the way the QC3 works is thus:
1. Wait for PD to timeout
2. Try and start QC
3. If QC starts (device voltage raises to 9V):
4. Use the QC3 pulses to raise the voltage to the user set voltage (9/12/20V)
5. If the pulses do not work, timeout and assume charger only supports 9V and lock at 9V.

You cant get into 9V mode with faulty hardware, as the same hardware is used to negotiate 9 and 12V.
Generally this hardware is tested at manufacture time as well.

So, questions for you:
1. Does the charger have the qualcomm licenced logo on it
2. Is it from a reputable brand that would have _actually done_ the testing
3. Did you remember to set the QC voltage to something higher than 9V in the menu?

JanezGroselj, 2021-07-18:
Thank you again for the thorough explanation! 
I think I understand the issue now. The TS80 works flawlessly with all QC chargers and powerbanks I've thrown at it, since it starts QC negotiation right out the bat. 
This makes the Pinecil a whole lot less QC compatible than the TS80 though, and I would think that is at least worth an asterisks on the spreadsheet compatibility comparison chart (which strongly influenced my decision to buy five Pinecils along with a few QC powerbanks to give as gifts of portable soldering irons, as I expected the checkmark for the QC3.0 on the spreadshet to mean the same thing on the Pinecil as it does on the TS80.) 

I really appreciate the effort in to this open source project and I'm thankful for all the hours spent behind the TS80 that were made more pleasant because of your firmware. 
Is there any way to make the Pinecil as QC friendly as the TS80? Can I manually turn off the PD in the code somehow, as I don't have a single PD power source, but a whole range of QC that I use with my TS80 (bench PSU, power bank, chargers...) along with super soft silicone USB-C cables which make handling the TS80 a dream.

And the answers:
1. At least one of the chargers has the QC3 logo on it (the 33W one I got two days ago with my new phone)
2. The above Xiaomi charger that comes with the phone
3. Yes, I set it to 20V. 

Thank you again for all the effort!

Ralim, 2021-07-18:
>Is there any way to make the Pinecil as QC friendly as the TS80?

You can recompile the firmware with the PD support turned off in the configuration file, this will prevent the PD code being compiled in / running.

If you have set the Pinecil to 20V, it is worth testing 12V, as some chargers instead of limiting at 12, will reset to 9 once you go beyond it (worth a try).

The Pinecil performance will be a lot lower on 12V. I **strongly** recommend picking up PD power adapters, as it is generally more compatible, not a proprietary standard  and where the future is going (QC3 is deprecated, later QC standards are PD under the hood).

Which table do you refer to?

JanezGroselj, 2021-07-18:
I have tried 12V and sadly it doesn't work :/

Do you think it is reasonable to put a PD-off setting in the menu? Otherwise presenting it as QC compatible is a bit misleading, if it doesn't work even with 33W qualcomm chargers. At least not in the context of being as QC compatible as the TS80.
I'm not very savvy with compiling, is there a guide somewhere or if it is perhaps super trivial to do for you, could I be so bold to ask you for a compiled .hex file? :)

This table:
https://github.com/Ralim/IronOS/wiki#differences-at-a-glance

> The Pinecil performance will be a lot lower on 12V.
You mean lower than on the TS80?
Is that because the tip is 8Ohm and on the TS80 it is 4Ohm?

Thank you again for all the helpful insights. :)

Ralim, 2021-07-18:
Its certainly reasonable to put it there, and will probably be done at some point.

It is 100% compatible to the QC _specifications_. Ergo, its compatible. Claiming its compatible does not promise compatibility with devices that do not match the specifications. It is as standards compatible as the TS80, given its the same code (just time shifted). If your charger chooses to not implement compatibility thats on the charger.

Yeah that table probably needs some footnotes, I've tried to avoid plastering a banner across the site that says to never buy a QC charger but sometimes I think about it :P 
Documentation is always a bit messy around this, as there is a tradeoff between saying its supported and saying its a good idea 😂 

So P=V^2/R, R is 4.5 for the TS80 tips and around 7.5-8.5 for the Hakko style tips the TS100/Pinecil use.
So at 12V, TS80 => 32W, where as the Hakko style tips are around 18W.
So a Pinecil at 12V is going to be closer to a TS80(P) at 9V. But the Pinecil also has a larger tip to heat up, so it feel feel _worse_ in some regards.

JanezGroselj, 2021-07-18:
Ok, so I've found out that you've been keeping some secrets from me :D I've noticed an action on master that has a PD timeout feature, which works perfectly for this case. :)
Though I suspect you've purposely withheld this information in order for there to be one less QC powered iron in the world. :D

And yes I wholly agree with you that QC is not the best idea for any of these irons. Or for the chargers. Thanks for clearing that up. :)
I'll be transitioning towards PD, and meanwhile using an old laptop brick to get those sweet 50 Watts.

So I think it's case closed here. Thank you again for the incredibly swift and helpful replies!! <3

Ralim, 2021-07-18:
Ah good catch, honestly, its less keeping secrets and more having the memory of a goldfish 😂

closed 2021-07-18 by JanezGroselj