Ralim/IronOS #1015: v2.15 on TS80 fried my Xiaomi 2C 20000mah powerbank

JanezGroselj, 2021-07-17:
**Describe the bug**
After updating the firmware on my TS80 and trying it on my powerbank it emitted a shortcircuit kind of noise and stopped working altogether. That was after only a couple of seconds of use.
I strongly suspect it was because of the firmware, since I have used this powerbank with my TS80 a lot and it seems too much of a coincidence for it to happen right after the firmware was updated...

**To Reproduce**
Not recommended actually, for obvious reasons,

**Expected behavior**
Not to have my powerbank killed.

**Details on your device:**
 - DeviceTS80
 - Release: [eg 2.15.40087E6]
 - Power adapter being used: Xiaomi 2C 20000mah powerbank

Ralim, 2021-07-17:
There is nothing the firmware can do to cause the hardware to consume more power than it already does.

The power consumed is a function of tip resistance, using ohms law.

The only thing it can really do is the pulse frequency but that has not changed recently at all.

Are you certain that you where only operating the QC within the specifications of the power bank?

What firmware were you on previously?

Can you share the exact specs of your power bank?

JanezGroselj, 2021-07-17:
Thank you again for the fast help and thorough explanations!

- "Are you certain that you where only operating the QC within the specifications of the power bank?"
I'm not sure, do you mean the power limit settings of the iron? I think that was either in the stock 24W setting or I changed it to off when I was doing my usual after-update setting adjustments.

- "What firmware were you on previously?"
I updated a few months ago before now, so 2.14 I think.

- "Can you share the exact specs of your power bank?"
9.0V @ 2.0A and 12.0V at 1.5A max 18W
Cell capacity: 20000mah 3.7V

Ralim, 2021-07-18:
So, if you had selected a 12V option when using this power bank, you would have been operating the power bank in overload.
Some power banks are designed with tolerances to be fine in this situation (My Cygnett one I opened up and its _just, maybe in spec_.) But most are _not_ and are _just_ rated for the 18W of power that they claim (even that can be.. pushing it)
You want a power supply rated for at least 33W+ to be able to use 12V on the TS80.

JanezGroselj, 2021-07-18:
Oh! So by selecting 12V and by the law of the Ohm (assuming a 4Ohm tip) 12V/4=3A, I tried to use more than 18W, whereas with 9V/4=2.25A, I would still be above 18W, but probably within tolerances (or even pushing it a bit at that)
At least it was a learning opportunity, thank you for doing the teaching :)
You can close this ticket, thank you again for the fast replies.

Ralim, 2021-07-18:
The tip _should_ be around 4.5 ohms, but its generally 4.3-5 ohms depending on batch.

The irons do not contain DC/DC converters, so all the iron does is turn the tip on or off and thats all it can do.
So when you do 12V you are pushing 2-3A which is beyond what the power bank can do :)

We have the power limiter option, but it limits _average_ power, not peak power. So it wont stop silicon going pop, but will work to reduce load on batteries for longevitiy. Its messy :(, but its sadly not trivial to change.

Still bad it died on you, not much I can do sadly though.

closed 2021-07-18 by Ralim