Ralim/IronOS #1021: [TS80P] Automatic display orientation the "wrong way up"

sigwinch28, 2021-07-21:
**Describe the bug**
When the "display orientation" is set to "A", the display is always "upside down" when I hold it in my hand, even after a reboot.

**To Reproduce**
Steps to reproduce the behavior:
1. Set the "Display Orientation" to A
1. Hold the iron in the left hand. The display will be "upside down".
1. Hold the iron in the right hand. The display will be "upside down".
1. Reboot the device. The error still occurs.

**Expected behavior**
I expected the display to be the "right way up" for the orientation I am holding it in when the "Display Orientation" setting is "A". For example, if I am holding in my left hand, then I expect the display to be the right way up

**Details on your device:**
 - Device: TS80P. PCB 2.
 - Release: v2.15.40087E6.
 - Power adapter being used: Apple Macbook USB-PD brick

Note you can grab these from the debug menu.

**Additional context**

https://user-images.githubusercontent.com/214649/126465554-5facb101-4efd-4270-ac66-69738c198ee7.MOV

sigwinch28, 2021-07-21:
Actually, the tip started glowing red hot so I suspect there might be something wrong with this iron.

iLLiac4, 2021-07-21:
Burned mosfet.

sigwinch28, 2021-07-21:
> Burned mosfet.

Straight out of the box? Oh :hankey:

iLLiac4, 2021-07-22:
Well it is not a complicated fix and is less than a dollar. It should be the same as for Pinecil or TS100. https://github.com/Ralim/IronOS/issues/934
https://www.youtube.com/watch?v=g-xXSJagz_A

iLLiac4, 2021-07-22:
From the pictures:
https://www.flickr.com/photos/eevblog/50111080862/in/photostream/
The mosfet for T80P version is 7409. Search for https://es.aliexpress.com/item/4000199115526.html 
To replace it you will need a heatgun.
Also make sure that the tip is ok since a shorted tip can kill the mosfet.

sigwinch28, 2021-07-22:
I changed the tip and everything appears okay now (except for the display orientation).
The new tip ramps up to 320C quickly and then the power draw drops to ~3W to keep the temp stable. Good stuff!

Ralim, 2021-07-23:
Display orientation being flipped is _odd_
We already flip and rotate the accelerometer for the TS80P compared to the TS80.

Just to be sure, you _did_ flash the right firmware?

sigwinch28, 2021-07-23:
I think so. Please double-check me:
[image]

sigwinch28, 2021-07-23:
Here is a video of the behaviour:
https://user-images.githubusercontent.com/214649/126769553-d4140f90-2163-4c90-b115-eb547dc5cec7.mov

discip, 2021-07-26:
Try this: https://github.com/Ralim/IronOS/releases/download/v2.15/TS80P.zip

sigwinch28, 2021-07-30:
> 
> 
> @sigwinch28
> Try this:
> [TS80P_EN.hex.txt](https://github.com/Ralim/IronOS/files/6881724/TS80P_EN.hex.txt)
> _**!rename first!**_

Thanks. I tried this, but it doesn't seem to make any difference. I will retry again later to make sure I didn't make a mistake.

discip, 2021-07-30:
Did you extract the hex file before  transferring it to the iron, or did you drag and drop it directly from the zip file?

sigwinch28, 2021-07-31:
Steps I followed:
1. Downloaded the `TS80P_EN.hex.txt` from your comment.
2. Renamed the `.hex.txt` to `.hex`.
3. Put TS80P in DFU mode
4. Dragged `.hex` onto USB drive.
5. Waited for the USB drive to come back with a `.RDY` extension.
6. Unplugged the TS80P.
7. Plugged back in.
8. Reset to factory settings.
9. Enabled "automatic" display orientation.

It's still upside down.

discip, 2021-07-31:
Good evening @sigwinch28,
you did everything as expected.

Are you able to open the device?
Here is a recommendation of how to achieve this: https://github.com/Ralim/IronOS/issues/630#issuecomment-809748658
Look for the version of the board and in the best case post a picture of the PCB.

Does it look like this?
[image]
or like this?
[image]

regards

sigwinch28, 2021-07-31:
I looked at the PCB and it looks like the bottom picture. The silk screen says TS80P V1.2.
[image]

discip, 2021-09-22:
@sigwinch28
Good morning,
is this issue still present with the latest build?

Shadowcrit, 2022-02-03:
I just wanted to add that I downloaded and flashed the TS80P artifact from https://github.com/Ralim/IronOS/actions/runs/1742279297 and the screen shows in the correct orientation.

https://user-images.githubusercontent.com/31373528/152269385-93f427c1-b9fd-42b5-a422-f101a9e13955.mp4

GrigoriOH, 2022-02-04:
> I just wanted to add that I downloaded and flashed the TS80P artifact from https://github.com/Ralim/IronOS/actions/runs/1742279297 and the screen shows in the correct orientation.

I downloaded the main, as well a this firmware and got the mismatched display orientation like OP in both cases. Just to be sure, I also tried the _EN and _DE version.

Thought it mitght have to do with either of those
> Fixing auto rotation bug in the LIS accelerometer in the TS80/TS80P
> Adds support for two new accelerometers
> -- SC7A20 (Future Pinecil batch) https://github.com/Ralim/IronOS/pull/786
> -- MSA301 (Newer TS80P) https://github.com/Ralim/IronOS/pull/761

so I tried different versions with the following results:
- works fine on 2.13
- goes upside-down from 2.14.1 onwards

I'd happily use 2.13, if it weren't for an issue  with my power supply, resulting in a wierd boot loop of the TS80P with 2.13.

https://user-images.githubusercontent.com/61632555/152551034-2f88f9cb-1aa9-4a9a-b47d-f66d475d19da.mp4

discip, 2022-02-04:
@sigwinch28
@Shadowcrit
@GrigoriOH
Would be helpful, if you would provide the model of your ACC using the debug menu.
To achieve that you need to hold down the button closest to the display *[button B]* and after the version number has popped up, push the button closest to the tip *[button A]* repeatedly until **`ACC . . .`** is displayed.)

Ralim, 2022-02-04:
I'm wondering if one model of accel has been flipped the other way up at some point in miniware's manufacturing.

If you could post your accelerometer version as well as roughly when you bought you iron we could try and narrow down whats going on

Shadowcrit, 2022-02-05:
Running v2.16.6054928 
ACC MSA301

I bought the Iron at most 1 year a go.

Ralim, 2022-02-06:
Do we think this is the MSA301 being mounted backwards to others; _or_ do we suspect MSA301 comes mounted two ways around?

Ralim, 2022-02-06:
@discip  what accel do you have and is it correct?

GrigoriOH, 2022-02-06:
@discip 
The ACC is shown as LIS2DH12 on version v2.16.72E0C17 - bought it on friday. Pictures of the board will follow.

discip, 2022-02-06:
@Ralim
> what accel do you have and is it correct?

### 1. LIS2DH12
### 2. It's correct.

@GrigoriOH
Very interesting.
At some point **miniware** changed the **ACC** to **_MSA301_** and increased the version to **1.2**.
Now your picture shows a **1.2** board and you say the **ACC** is **_LIS2DH12_**. 🤷‍♂️😵

Ralim, 2022-02-06:
Ah damn,
I'm guessing there is a way their firmware is detecting the newer board rev and flipping it.
Not sure if its worth tracing every track to find differences or if we add a flip option to the menu 😰 

Would you be ok with just having an "invert orientation" toggle in advanced settings ?

discip, 2022-02-06:
If you ask me, it would be nice and clean if we could keep the *extra entries* as low as possible by detecting the boards version (If that is possible.).

Ralim, 2022-02-08:
Aha my question to Miniware was answered 🙏🏼 

Looks like they have started using the SC7 as well as the LIS.
SC7 wasn't enabled in Miniware builds until now so we were using the LIS driver.
... So

Could you please test this build : https://github.com/Ralim/IronOS/actions/runs/1811536243

And then report back what model your accelerometer shows up as ?
Also if orientation is still backwards or not :)

discip, 2022-02-08:
iron  |  Firmware Version  |  ACC  |  correct orientation
-|-|-|-
**TS80P**|&emsp;&emsp;**https://github.com/Ralim/IronOS/commit/641fc76e4980b0e4ec61f1e095bc0aa0120f3c7a**|**LIS2DH12**|&emsp;&emsp;&emsp; **&#10004;**

GrigoriOH, 2022-02-08:
I was finally able to try the new version - unfortunately, the display is still upside down.
Accelerometer is also still shown as LIS2DH12 on v2.16.FCE7DA6.

Anything more I can do for you?

Ralim, 2022-02-09:
Damn, was hoping that one was the model.
Could you give this action build a check and see what model shows up?
https://github.com/Ralim/IronOS/actions/runs/1817624613
If its still the LIS, might write some code to dump out the common id registers to let us sanity check.

GrigoriOH, 2022-02-09:
Sure, I'll do whatever you need ;)
It still lists the ACC as LIS2DH12.

I am supposed to download the TS80P build unter artifacts, am I? Just to be sure I don't do anything wrong, delivering faulty information.

Ralim, 2022-02-09:
Hmm damn.

Yep that's right, and your version you commented last time proved you did  it right :) 

Not sure then what's the go, talking to miniware they haven't added a new one I don't think  but will go ask again to check what the go is.

Apologies on the delay, trying to rule out what I can before the worst case solution (manual invert in the menu)

Ralim, 2022-02-11:
[TS80.zip](https://github.com/Ralim/IronOS/files/8052170/TS80.zip)

Hia,
If you get a chance could you flash this and try it.
Its a super minimal firmware that jumps some numbers to the screen.
It should show one number per line of the screen, and then when you press a button show another set and so on, until the screen goes blank.

If you could record out those numbers here it would be great.
Afterwards, just flash back to a normal software release for iron functionality.

For example you might get
```
084
128
```
Then you press the button and get
```
096
123
```

This code is just dumping out exactly what ID's on the I2C bus are there, and reading one of the registers from them to let us try to match up with datasheets 🤞🏼

GrigoriOH, 2022-02-11:
Hey Ralim,
no need for any apologies, I am sorry for creating trouble on the dev front ;)

Here are the addresses you asked for:
```
025
017
---
060
003
```

Ralim, 2022-02-12:
@GrigoriOH 
Could you try the release from here : https://github.com/Ralim/IronOS/actions/runs/1832151266

And check (A) which it ID's your iron as and (b) if orientation is flipped for you still :)

GrigoriOH, 2022-02-12:
Okay, just installed v2.16.90431CD and got the SC7A20 to show up 🎉 
Unfortunately, the display is still flipped upside down.

closed 2022-02-12 by Ralim

Firebie, 2022-02-12:
Sorry, but latest master on pinecil gives 'No accelerometer detected'.

Ralim, 2022-02-12:
What accelerometer does your firmware have on previous releases?

Firebie, 2022-02-12:
on stock 2.16: BMA223

Ralim, 2022-02-12:
My BMA223 two units here are working fine _hrmm_.
Do you see the accelerometer model show up in the debug menu?
Was your Pinecil bought from the Pine64 store?

Firebie, 2022-02-12:
I bought it from official Pine64 store
2.17 shows no accell
2.16 shows BMA223

Firebie, 2022-02-12:
If I comment #define ACCEL_SC7 in pinecil config, BMA233 is back again

Ralim, 2022-02-12:
Looks like the BMA isn't liking the probe of register 0x0F _sigh_.
Workaround is simple enough to rebuild with order flipped for probing.
Interesting doesnt show for me.
Can you test this build
[Uploading Pinecil_EN.dfu.zip…]()

Firebie, 2022-02-12:
Can I upload .dfu using "GD32 MCU Dfu Tool.exe"? I used only to upload .hex with it.

Ralim, 2022-02-12:
[Pinecil_EN.zip](https://github.com/Ralim/IronOS/files/8052987/Pinecil_EN.zip)

Firebie, 2022-02-12:
Yes, your latest works

PR #1225 (2022-02-12): TS80P Accelerometer version detection again
Trying to fix #1021 again