Ralim/IronOS #1023: Can't flash the newest 2.15 EN firmware

cewbdex, 2021-07-23:
**Describe the bug**
Can't flash the 2.15 TS80_EN.hex, any tried other language and they work (CS, DE).

**Details on your device:**
 - Device: TS80
 - Release: 2.15

[TS80_EN.ERR.txt](https://github.com/Ralim/IronOS/files/6867835/TS80_EN.ERR.txt)

ricktendo, 2021-10-11:
Try flashing in Windows 7, or use the script in Linux

https://github.com/Ralim/IronOS/tree/master/Flashing

`./flash_ts100_linux.sh <hexfile>.hex`

cewbdex, 2021-11-02:
I have tried on both Linux and Windows, my DFU is 3.45, is there any newer? Will try on another computer too.

Devnol, 2021-11-20:
Similar issue in macOS, it flashes but only like 10% of the time.

Ralim, 2021-11-20:
There isn't a whole lot that we can really do.
If you are technically minded and know what USB-DFU is, you can try loading the alternative bootloader that is over [here](https://github.com/eDesignOSS/ts100-bl-flasher/releases). But if you dont know what your doing that may be more annoying as that will require you to use dfu-util for all flashing.

Otherwise only thing you can do is try and get the manufacturer to care enough to either fix or open source their bootloader.

Not really in the scope of this firmware sadly, and writing a re-implementation of their bootloader feels like a nightmare

Devnol, 2021-11-20:
Yeah I see, I was just wondering if there is any known solution to this or whether you know exactly what is causing it to fail and how to mitigate it. All I see reading the shell script is that you're manually mounting and unmounting the drive and using cp to copy the file, nothing I would consider special.

Ralim, 2021-11-20:
The main thing as far as I know (still a guess) is that the OS must write every sector _in order_.
_any_ reshuffling of the data write order appears to break it.

So a lot of the "magic" that works for people is often using older and dumber drivers to mount with.

I do suspect some careful `dd` might work well but untested

Devnol, 2021-11-20:
Hmm, but the weird thing is that flashing works sometimes and it fails others

Devnol, 2021-11-20:
disk destroyer sounds like a good idea, I'll report back in a little

Ralim, 2021-11-20:
I believe that is down to the host OS; most likely depending on how things are being buffered

Devnol, 2021-11-20:
dd'ing with a bs of 1b and no other special flags also yielded in an error so it appears it's something very low level

Ralim, 2021-11-20:
I'm not really sure tbqh.

I could try capturing some logs of the bus but i dont have the time to really dig into that anytime soon :(

Devnol, 2021-11-20:
Alright, thanks for your time anyway, if I do find anything interesting I'll report back. One last thing is do you remember the syntax needed for dfu-util to flash to the iron?

Devnol, 2022-04-01:
Oh, sorry this was left stale, iirc I got it working a little later by flashing the miniware fw and then ironos, which somehow fixed it.

closed 2022-04-01 by Ralim

cewbdex, 2022-04-02:
Flashing 2.17 EN works, so the can't fix can be removed too.

Ralim, 2022-04-02:
Can't fix is because the root cause of this issue is not something I can really fix