Ralim/IronOS #1027: TS100 weird temperature readings

jeanblanchard, 2021-08-13:
**Describe the bug**

I'm getting weird standby temperature readings on my TS100, off by 100+°C.

What's event weirder is that the temperature reading varies considerably based on the input voltage (I have a variable output power supply).

In my ~25°C room, I get __120°C__ with 12V input, and up to __264°C__ with 24V input.

This is unaffected by calibration, or the installed tip.

**Details on your device:**
 - Device: TS100
 - Tips: TS-I, TS-B2, TS-BC2
 - Release: 2.15
 - Power adapter being used: variable output power brick (JDT-001)

**Debug menu output**
At 12V:
```
RTip 845
CTip 120
CHan 177
Vin 12.0
PCB 1
```

At 24V:
```
RTip 845
CTip 164
CHan 618
VIn 24.0
```

When Plugged in to USB:
```
RTip 805
CTip 53
CHan 520
VIn 0.0
```

**Original Firmware**
The original firmware (V2.18) reports a temperature that is not the most accurate either, but not as crazy, showing something like 42°C to 45°C

Ralim, 2021-08-13:
This is interesting,
Your in-handle temperature sensor for the cold junction temp is reading rather odd.
I wonder if they have swapped components _again_.
Would it be possible for you to remove the back cover and try and take a photo of the pcb inside? (dont remove the pcb).
There should be a small black part between the copper pieces that poke up to connect to the tip, and I would like to check (a) its fitted and (b) if its the expected part of if the board design has changed.

Is your unit a miniware official one or a rebrand?

jeanblanchard, 2021-09-03:
Hi, sorry for the delay, didn't see your reply.

My TS100 is actually quite old, bought in October 2017. I just wasn't aware of alternative firmwares until recently...

Anyway here is a picture of the pcb
[image]

I believe it's an official Miniware, it does have the sticker (and it's probably all there was at the time)

River-Mochi, 2023-03-24:
This  ticket hasn't been looked at in a while but perhaps try something we do on the Pinecil iron since it uses the same cartridge tips.    wipe the white  rear end of the  cartridge very well with isopropal alcohol (IPA).  it's worth a try as it resolved a lot of weird  Raw tip values and  fluctuating  temperatures on Pinecils especially with brand new  cartridge tips (I assume factory oil/residue on them). 

clean at the  +/-  but  I wipe whole thing.  
[image]