Ralim/IronOS #1029: can't upgrade ts100

ludov1c, 2021-08-20:
hello i can't upgrade my new ts100 ,
enter in DPU mode : 3.45 , but when i drag and drop the .hex 
[image]
the file .hex is too long and the ts100 disconnect

Ralim, 2021-08-20:
Hello,

Can you confirm that at the time of trying this, the screen is showing "DFU"?
Can you confirm you have extracted the file before attempting to copy it to the drive?
Can you confirm you have no anti-virus that supports file scanning on access turned on?
Can you confirm you have tried the double drag and drop method as mentioned [here](https://github.com/Ralim/IronOS/blob/master/Documentation/Flashing.md#miniware-the-file-is-showing-up-with-the-extension-err)?

pellz0r, 2021-08-21:
I've had this issue as well. Plugging it in another USB port solved it for me.

ludov1c, 2021-08-21:
i can upgrade only with the MINI218.HEX official, i will try  with no anti-virus

Ralim, 2021-08-22:
Please also try with the English IronOS firmware release as its the most tested.

discip, 2021-09-21:
@ludov1c
Good evening Sir,
is this solved?
If so, please close this issue.

thanks in advance

closed 2021-09-25 by Ralim