Ralim/IronOS #1033: TS100 No any USB activity and temperature problem

NikolayMurha, 2021-08-31:
Hello. 

I have a complex problem. One day I turned on my TS100 with IronOS and saw the temperature jumps.  First of all, I decided to update the firmware, and was surprised to  find that the system did not react in any way to the connection (I try  on mac, windows and raspbian) - no any activity, no any new hardware on  Win, no any new dmesg messages on linux/mac.
I took TS100 with original firmware from a friend (hereinafter Original TS100) I swapped the boards and have to the  conclusions:

- USB connection does not work (broken) on my MCU - USB connection with  from Original TS100 MCU worked on both motherboards. My MC can't connect  on both motherboards. I understand correctly that regardless of the type of bootloader / firmware, there must be at least some USB activity? Two options - poor contact or the interface burned out.

- Temperature sensor broken on my motherboard - MCU from a working soldering iron also has temperature problems on my motherboard.

Thus, I have two cases. My guess is static electricity. Otherwise, I don’t understand why several circuits failed?

I have ST-Link but I have never worked with it. I found how to connect it, but maybe you have a short instruction on which firmware file to upload, and most importantly, how?

On thecond issue, I guess that potential cause in U1 or U6 (Q1 is intact). Can you tell me the steps to diagnose these circuits? There is a multimeter available and DSO 150 oscillograph :). I'll provide a debug menu info later. 

Thanks.

paulfertser, 2021-08-31:
Hi Nikolay,

On Tue, Aug 31, 2021 at 01:40:24PM +0000, Nikolay Murha wrote:
>   • USB connection does not work (broken) on my MCU - USB connection with from
>     Original TS100 MCU worked on both motherboards.

Please take into account IronOS doesn't include a USB stack so you
should be staying in the bootloader to see any USB activity.

NikolayMurha, 2021-08-31:
@paulfertser thanks for respond. 
> Please take into account IronOS doesn't include a USB stack so you should be staying in the bootloader to see any USB activity.

What you mean "staying in the bootloader"? How to enter to the bootloader?
If I understood you correctly, when IronOS flashed, then mass storage device wouldnt exist? And firmware upgrade process different than on stock firmware?

paulfertser, 2021-08-31:
On Tue, Aug 31, 2021 at 08:45:05AM -0700, Nikolay Murha wrote:
> What you mean "staying in the bootloader"? How to enter to the bootloader?
> If I understood you correctly, when IronOS flashed, then mass storage device
> wouldnt exist? And firmware upgrade process different than on stock firmware?

Flashing IronOS doesn't change the bootloader. Upgrading process
should be identical: hold the button closer to tip, connect
power. Miniware bootloader would init the display and confusingly show
"DFU" on it (when in fact it is not related to DFU and is visible as
an MSC device on the host). You should be checking if USB works in
this mode only.

NikolayMurha, 2021-08-31:
@paulfertser Oh, sorry! Of course, I forgot about hold button! 
After connect iron with stock firmware I have storage for config, and I expected a similar reaction from mine. 
The day before, I was updating the firmware on my 3D printer, and I just copied it to the SD card. All this information was mixed in my head, and I was thinking wrong. 
Oh, and I was hoping to dive deeper into the STM controller. Another time: D 
Anyway this is great, and now there is one problem less.

NikolayMurha, 2021-08-31:
I think, I fixed the temperature problem. From the beginning, I washed the contacts and the board with alcohol.  After that, the temperature became maximum, as if a tip had not been inserted.  I soldered the U1 pins and the temperature reading came back and became stable

Ralim, 2021-08-31:
Hello, 

I think this is mostly sorted...I think.

But to confirm:
IronOS has no USB interface enabled, as it's entirely superceded by the on device menu.

You will have to hold the button at boot for the miniware bootloader to run, you can confirm this is running by seeing the DFU text on the screen.

The miniware bootloader is required as the STM32F103 does **not** have a USB bootloader built in.

If you did have to recover a completely blanked unit, that is what this file is in the repo for https://github.com/Ralim/IronOS/blob/master/Development%20Resources/DO%20NOT%20FLASH_Flash%20Backup.hex

closed 2021-09-25 by Ralim