Ralim/IronOS #1037: Feature Request: Drop Detection

dburr, 2021-09-04:
**Is your feature request related to a problem? Please describe.**
So I was soldering today and I fumbled and dropped my iron. Of course in the act of reaching down to pick it up, my foot happened to brush against it. Ouch.

**Describe the solution you'd like**
Don't these irons have accelerometers in them? Would be nice if the iron could detect extremely violent motions (i.e. dropping them) and shut themselves off.

**Describe alternatives you've considered**
Not being clumsy... easier said than done tho

**Additional context**
Could save a lot of people from accidentally burning their feet (or worse, burning the carpet and maybe starting a fire)

paulfertser, 2021-09-06:
Hey Donald,

On Fri, Sep 03, 2021 at 08:13:10PM -0700, Donald Burr wrote:
> Additional context
> Could save a lot of people from accidentally burning their feet (or worse,
> burning the carpet and maybe starting a fire)

I propose an experiment: heat the soldering iron up to the temperature
you're usually using, then disconnect it from power altogether and
drop immediately on the carpet and/or feet :)

My point is that since the iron can't immediately cool down anyway
would the added code complexity and possible annoying false positives
be worth it?

discip, 2021-09-06:
Actually this has been brought up at least twice in the past:

https://github.com/Ralim/IronOS/issues/314
https://github.com/Ralim/IronOS/issues/625

As mentioned by @Ralim himself and others this is not worth the effort.
### Would be nice, if at least the last ones would be closed.

Ralim, 2021-09-07:
Yeah this is absolutely a duplicate and its never really been done as its benefit is quite small compared to both the complexity of detection and the tuning required to avoid false positives.

Going to close this and just keep the original one

closed 2021-09-07 by Ralim