Ralim/IronOS #1042: QC3 Continuous Mode

D33pWat3r, 2021-09-14:
I tried the IronOS firmware on my TS80P for the first time.
In the source code I saw that the QuickCharge3 Continuous Mode is supported. But how can I use this to trigger a voltage higher than 12V.
The TS80(P) tips have a resistance of 4.5Ohms and the max. current of QC is 3A. This would allow a maximum voltage of 13.5V with 40.5W.
All of my QC3 chargers support up to 3A and that would be a great upgrade.

Ralim, 2021-09-14:
Hia,
At the moment its locked to 9/12/20V options (model depending).
Partly as the accuracy isnt great in the QC3 0.2V steps, and partly as the TS80(P) are not _ideally_ thermally designed to be pushed very far.

If you wanted to test this I can spin you abuild with say 13.4 as the setpoint for you to try, but if it breaks your iron that would be on you ?

D33pWat3r, 2021-09-14:
A test build with 13.4V (39,9W) sounds good. Can you also change the setting "#define MAX_POWER_LIMIT" to 40W?
The risk of damage is my problem :)

I've already tried out the original firmware of the TS80P with 13V, 13,5V and 14V and it worked very well.

In the future I will hopefully set up the development enviroment and do a lot of tests with QC3.

Ralim, 2021-09-14:
Yep I'll get you a year build tomorrow (Aussie  time here).

Will look to add this adjustment to the menu too for you when time permits

Ralim, 2021-09-15:
[Downloads.zip](https://github.com/Ralim/IronOS/files/7169571/Downloads.zip)
Hia, dont have a >12V QC unit near me to test, have done a fairly blind guess at the above.
Will try and get onto giving you a menu for this as soon as I can though (this week)

D33pWat3r, 2021-09-17:
Hi, I test your build and it works:
[image]
My TS80P has successfully triggered the 13V.

I did some tests to measure the possible performance (the soldering tip is heated in water):
[image]
The max measured power is 38W (peak)

Thanks again for your build :)

Ralim, 2021-09-17:
Ah that is most excellent, glad it worked 😅 

Did a pass of making it show up in the menu.
If you feel like having a look at #1046 (currently its latest build is here https://github.com/Ralim/IronOS/actions/runs/1245320478 )

Would love to know your thoughts / if it works for you etc :)

D33pWat3r, 2021-09-17:
I tested your latest build and the ability to adjust the voltage via the menu is a great thing 🙌

but I found two things that are still buggy:
- the max. triggered voltage is 13V, but in the menu you can set up to 14V
I think the following lines (61-66) of code in "source/Core/Src/QC3.cpp" are the problem:
```
#ifdef POW_QC_20V
  if (Vx10 > 200)
    Vx10 = 200; // Cap max value at 20V
#else
  if (Vx10 > 130)
    Vx10 = 130; // Cap max value at 13V`
```
   I'm not sure, but I think the macro "QC_VOLTAGE_MAX" would be better than the static values

- The choice of voltage in the menu is also a little buggy. Only every second voltage value is triggered.
The behavior is probably due to the following lines (80-81) code in "source/Core/Src/QC3.cpp"
```
    if (steps > -2 && steps < 2)
      return; // dont bother with small steps
```
Are there problems with single 0.2V steps?

Ralim, 2021-09-17:
Ah yep can fix both of those in the morning.

The first is just I forgot about that check

The second was originally to stop it hunting back and forth but it's probably not really required tbh

Ralim, 2021-09-18:
@D33pWat3r 

New build available at https://github.com/Ralim/IronOS/actions/runs/1247683469

_should_ fix those issues I hope :)

D33pWat3r, 2021-09-18:
hi, your new build fixed my last issues 👍
It's pretty cool that the TS80 (P) now has a power upgrade 🔥

But I noticed some little things:
- When the voltage is triggered up to the desired value, there are some overshoots
In the picture you can see that the voltage overshoot is ~14,6V.  
But i think it's not a problem.

- In the range of 38-40W output power there is a gap of the power between my USB-Power Meter (~40W) and power shown on the display (~37W).
If I have it right in my head, the TS80(P) has no direct way of measuring the power.

So I am happy with the current QC3 modus :)
[image]

Ralim, 2021-09-18:
Excellent

(1) this is sort of by design of the dumb what it's doing it (seeking rather than calculating).
This is done so that if your cable droops under load it tries to maintain voltage at the TS80(P)

(2) You are correct, the iron can't measure the current so it's a guess based on tip resistance. It also ignores the power used for the iron control (micro, OLED, accelerometer) and the voltage regulator for them. Do it's generally close but never perfect

closed 2021-09-25 by Ralim