Ralim/IronOS #1050: Please include a DFU suffix in firmware files for USB DFU

tormodvolden, 2021-09-19:
This is recommended (see also the standard https://www.usb.org/sites/default/files/DFU_1.1.pdf) so that the file includes a checksum, plus USB VID/PID so that users don't burn firmware files onto the wrong device (there are for instance laptop built-in bluetooth adapters with DFU support, and nobody wants to brick those).

For instance the tool dfu-suffix from the dfu-util package can be used to easily add such a suffix.

Ralim, 2021-09-20:
Yeah makes sense.
Will need to have a look into the exact incantation to add this for the Pinecil / Miniware irons to create dfu files instead of .bin's

tormodvolden, 2021-09-20:
Something like `dfu-suffix --vid 28E9 --pid 0189 --add Pinecil_xy.bin`

PR #1058 (2021-09-25): Generate .dfu files
Adding in dfu files that have the dfu suffix

Will drop bin files once these are known to work.

Fixes #1050
closed 2021-09-25 by Ralim