Ralim/IronOS #1055: 'scrollbar' not behaving right

discip, 2021-09-21:
**Describe the bug**
The 'scrollbar' is in some registers not using the whole area it should, hence its length is not right either.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the main menu and go through it not entering any submenu (keep pressing the **`B-button`**).
2. Watch the 'scrollbar' making a jump between the 3rd and 4th entry and not going all the way down to the bottom, when the last entry is reached.
3. Do this for the **`Advanced settings`** as well.
4. Similar outcome!

**Expected behavior**
1. The length of the 'scrollbar' should be a fraction of the height of the display defined by the amount of entries.
2. The 'scrollbar' should move according to the ratio mentioned above.

**Details on your device:**
iron  |  Firmware Version  |  PCB Version  |  Power Supply
-|-|-|-
**TS80P**|**https://github.com/Ralim/IronOS/commit/5729479c87910146936a822d564957965e4620f8**  |**2**|**[UGREEN 65W GaN Charger](https://www.ugreen.com/products/65w-gan-usb-c-fast-charger?variant=32127034654782)**

**Additional context**
I think this behavior is related to the fact that there is
1. one more, not yet activated, entry for the **`main menu`** and
2. two hidden entries in **`Advanced settings`**, if **`Power pulse`** is set to **`Off`**.

sandmanRO, 2021-09-22:
Hello @discip, this one is not going to be so straightforward to solve. Indeed, the issue is related to the dynamic menus, that could show or not depending on other settings. The issue with the current implementation is that it requires to paint (in background) a menu to find out if it should be painted or not in foreground. Paining a menu, in background or not, is computationally expensive so perhaps it is not the most efficient way of handling this issue. What I did in my implementation was to add a function prototype to the menu structure that would address just this question: should that menu be painted or not? With most of the menus, the method was assigned to a function that always returns true. With other menus, the assigned function would do that specific menu simple logic and return true or false. If true, the menus is to painted, or otherwise returns false. However, every time some settings change you have to revisit all the submenus in that menu and see how many menus should be painted, and (re)compute the cursor size and position and the number of submenu entries pertaining to that menu.  Luckily there are way to do that on the fly with the minimum computational effort possible. 
P.S. If some people wonder why I keep talking about computational efficiency, you have to keep in mind that Ralim implementation refreshes the screen at the impressive rate of 20 times per second so all the computation required to paint one screen would be repeated 20 per second.

discip, 2021-09-23:
@PixelPirate
Good evening,
since you brought the scrollbar in this project to life [release v2.10.1](https://github.com/Ralim/IronOS/releases/tag/v2.10.1), I think you should have a look into this. PLEASE 🥺

@paulfertser
As you have shown you skills in *'bug hunting'* by fixing a similar issue in the past https://github.com/Ralim/IronOS/commit/099d6b8aee4ae69eb585f13bfdd28813af481985, I think this might be another case for you. 😃 Please do not refuse.

thanks to both of you in advance
kind regards

Ralim, 2021-09-24:
@sandmanRO is on point here, the best way I know of to do this is to pre-calculate the counter ahead of render time, and re-evaluate it whenever an option is changed.

I can probably try and get this done over this weekend pending nothing else urgent stealing my attention

PixelPirate, 2021-09-24:
Hi, I'll try go get some time to look into this. I think all needed information is available at the location where the scrollbar is calculated, so a fix should not be too difficult. Maybe it's just an off-by-one-error kind of thing.

sandmanRO, 2021-09-24:
Hello Ralim, when you will have time to look into this, may I also suggest you consider a small cosmetic change to the scroll bar? My first reaction to the current implementation was I genuinely thought that the one dot width segment at the very right of the display must be some graphic bug. It was only afterwards I realized that in fact that is scroll cursor. Nevertheless, in my build I have decided to address this by adding a dotted line (like in the picture below) so the resemblance to a scroll bar would let no room for doubt. But then again, I'm an old guy and my eyes are no longer what they used to be. Maybe for the young sharp eyes that one dot segment screams scroll bar already and what I did is actually overkill.
[image]

discip, 2021-09-24:
@sandmanRO
Good evening,
I do not know, if I am the only one with this opinion, but I'd rather have a **tinier scrollbar**, than **less room for the text** on the screen. 😊
In some languages this might be not a problem, but in others every single pixel is crucial, because of the expressions being relatively long.

sandmanRO, 2021-09-24:
Hello @discip, sure thing...every aspect should be considered. Although using 2 pixels vs. one pixel wide scroll bar might not sound much, I can imagine that there are languages where you just do not have that 1 extra pixel available. However, following the same logic, there might be situations where you have no room for scroll bar at all so in my implementation the scroll bar can be turned on/off from the display options. But this is not the real reason I did it. I have colleagues that thought the scroll bar would be useful while others preferred a "cleaner" display so I tried to make everybody happy.

discip, 2021-09-25:
@Ralim
# Thank you Sir! 🤝❤👍

closed 2021-09-25 by discip