Ralim/IronOS #1062: debug menu printing DC, QC or PD instead of 0, 1, 2

discip, 2021-09-25:
@Ralim
Sorry to bother you again,
but since you changed **`PCB`** to **`ACC`**.
I Think for consistency it would be great, if **`PWR`** would also not be encrypted printing a number, but **`DC`**, **`QC`** or **`PD`** instead.

I looked into **`GUIThread.cpp`** but realized, there are other things going on than with formerly **`PCB`**.
So, may I humbly ask, if you would do this as well? 🥺

Ralim, 2021-09-25:
Ah yeah fair call
Probably late tonight / tomorrow :)

discip, 2021-09-25:
Looking forward to this, and definitively willing to test! 😁

Ralim, 2021-09-25:
There you go :)
Looks to work for me but will wait for your test as I dont have a QC source handy

discip, 2021-09-25:
@Ralim
Thank you! 👍
Works for me as well!

PR #1074 (2021-09-25): Ralim/named power source

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [x] There are no breaking changes

* **What kind of change does this PR introduce?**

Moving to human friendly names for the power source for @discip 
Closes #1062 

* **What is the current behavior?**

* **What is the new behavior (if this is a feature change)?**

* **Other information**:
closed 2021-09-25 by discip