Ralim/IronOS #1078: Brightness/Invert icons

federicodilo, 2021-09-26:
As said in https://github.com/Ralim/IronOS/pull/1010#issuecomment-917598230 I think there's no need to break continuity with other menu entries by using icons just for two options.

Maybe a complete set of icons would be best suitable for a "graphical" version of the menu?

discip, 2021-09-27:
> About brightness control: a fake scale from 0 to 9 (or 1 to 10) would in case increase the available space for the entry (and maybe make more sense for the user).

Done https://github.com/Ralim/IronOS/pull/1083. 😊

sandmanRO, 2021-09-28:
Hello Gentlemen, I see that you decided to implement the color invert. I did it too at some point but only because I was kind of curious to see how it could be done, just to disable it as soon as I saw the outcome. It looks bad if you ask me. If there would be a white (or at least light color) and perfectly positioned mask over the OLED display, which is not the case, the inverted color scheme would look ok but as it is, i.e. the fonts are printed right from the top raw of the OLED, they would look sort of unfinished as they would blend on the top side with the OLED's black border. If there would be a white one pixel top border the things would probably look differently but then again, on small / double raw text, there is no extra vertical pixel available unless you decide to go with one pixel smaller fonts.
Regrading the brightness (contrast), although the hardware supports a value within 0...255 (it is worth stating that 0 is not completely dark but a minimum hardware internally defined contrast) in my implementation I put a top limit of 100 on this. Increasing the value farther towards 255 would not lead to a significant extra gain in the actual contrast but it would put an unnecessary strain on the organic molecules within the OLED. Speaking of the OLED life expectancy, I know that there are all sort of speculations about and I don't want to address those. The main 3 factors that kill an OLED are the internal Joule heating, external UV (ultra-violet) light, and the (most often ignored) imperfections of the sealing that in time leads to the OLED's organic layer oxidation. These are my two cents thoughts on brightness / invert color.

discip, 2021-09-28:
@sandmanRO
Good afternoon.
If it was up to me, I would not have implemented the *color invert option* at all, due to almost exactly the same reason you mentioned. 😃
Since @alvinhochun is the creator of both the *color invert* & the *brightness adjustment* features https://github.com/Ralim/IronOS/pull/1010, I did not know, that it was possible to go as low as **`0`** and that it was not that good to go all the way up to **`255`**.
Thank you for pointing that out! 👍
Will do some more testing and adjusting. 😁

sandmanRO, 2021-09-28:
Hello discip,
This is one of those things that until you implement it's kind of hard to foreseen the outcome. I personally try all sort of things mostly because I'm curious to see how they could be done but sometimes a feature, although appealing at the first sight, might lead to other negative effects so you have to decide if it's worth keeping it or not.

Ralim, 2021-09-29:
I agree on both of these fronts, I did try inversion a long time ago and never really liked it :/
But also, the cost of adding the option is very low, so can put it in and leave it there so everyone can satisfy their curiosity. Maybe someone will think of a design that does work well 🤷🏼 

OLED's are a pain in terms of longevity but I have found storing out of sunlight in "kind" conditions they last quite a long time.

sandmanRO, 2021-09-29:
Hello Ralim, I think you are the wise one here. Seeing is believing... maybe other people were wondering how it would look like and now they have it....maybe, as you said, someone with fresh eyes will come up with an idea that will get this right in a surprising way.

Ralim, 2021-09-29:
I've been proven wrong enough times now that I just want to let people choose and then wait and see 😅 wisdom hard earned perhaps 🙂

federicodilo, 2021-12-02:
Seems to me that an outline of 1px (1px spacing) enanches colour inversion's readability.
Here's a sample: [image]

_(It reminds me the good old times with GBC/GBA)_

closed 2022-04-24 by federicodilo