Ralim/IronOS #1092: Please, add descriptions for translators

playday3008, 2021-09-29:
Like describe expected behavior after change some setting, etc.

Devnol, 2021-11-20:
I second this, also provide some context as some languages have multiple translations for one word

Ralim, 2021-11-20:
How do you want this to be conveyed?
We could expand out the html editor and add descriptions in the defs file?

Devnol, 2021-11-20:
I don't have any particular way in mind but ideally yeah either a description of what exactly is the context of the string we're translating, where it can be found in the menus and stuff by adding a description field the same way there's a length field I guess.

Ralim, 2022-01-18:
@playday3008  and @Devnol 
I've added descriptions to the defs file in #1183, if you feel like giving them  a pass to see if they are helpful / what I've missed.

Also creating a script to auto-gen some docs for the repo on a settings break down that is open to criticisms.

Ralim, 2022-02-04:
Going to consider this covered now we have helptext in the editor (and it will also show up in the docs folder)

closed 2022-02-04 by Ralim