Ralim/IronOS #1100: Baseus PSU

Kernex7, 2021-10-02:
Hello, do you know if this is normal, or if it is better to change your diet? I don't have this with another

Uploading iykzzl.mp4…

Ralim, 2021-10-02:
Hello,
Can you please detail which firmware you are running; and the other information that is in the issue template please :)

Kernex7, 2021-10-02:
Yes sorry

**Describe the bug**
As in the video the Pinecil does some kind of restart which I don't have with other chargers, I am using a Baseus GAN 65W Lite 

**To Reproduce**
Just plug the pinecil

**Details on your device:**
 - Device: Pinecil
 - Release: Latest commit in the actions [here](https://github.com/Ralim/IronOS/actions/runs/1298131575)
 - Power adapter being used: Baseus GAN 65W Lite

Ralim, 2021-10-02:
So the restart is a PD timeout most likely. Nothing will break, and it looks to be working on the second attempt.
Without access to a PD datalogger or analyser its hard to know _exactly_ why the negotiation failed.

If its working then dont worry and your fine. I use a very similar PSU as my portable unit here, which I can't reproduce this on sadly.

Also note that its not the "latest" action of the master branch, but instead its from a PR.
(I dont expect that to matter at all as Discp keeps in sync but just for the sake of noting it down).

If you have time (your call) would love if you could compare to the latest RC1 release since I've just finished a pass of stuff on PD and would be interested if it occurs on the older stack or not

Kernex7, 2021-10-03:
If it does not damage the screen or the device, like short etc, then it is not a prob
I have a USB voltmeter taking measurements but I don't know if it's enough lol

I flashed the RC1, sorry for the quality, but on the right we see the volts go up 1 by 1

https://user-images.githubusercontent.com/91785605/135749283-c856a74b-fad5-4d72-8e8b-715ed3893bd2.mp4

Johnex, 2021-10-03:
@Kernex7 We found this issue a few days ago and it's fixed here:

https://github.com/Ralim/IronOS/issues/1070

Download the files from here to test:

https://github.com/Ralim/IronOS/actions/runs/1295219064

discip, 2022-03-31:
@Kernex7
Is this still happening to you running the lastest commit?

If not, please consider to close this issue. 😊

thanks

discip, 2022-04-24:
@Ralim 
As @Johnex pointed out this is similar to #1070, which is already solved. So this should be closed as well. 😊

closed 2022-05-05 by Ralim