Ralim/IronOS #1111: Accelerometer not found

CvRXX, 2021-11-10:
**Describe the bug**
When I start the soldering iron it shows "accelerometer not found". The sensitivity is automatically set to zero in the settings. 

**To Reproduce**
1. Turn on soldering iron. 
2. See the message
3. Set sensitivity to 7. Power cycle. It's back at 0.
4. Set sensitivity to 7, turn on the heating. The timer will count down regardless of movement.

**Expected behaviour**
A working accelerometer so the message should not show. The sensitivity should remain at 7 and the timer shouldn't countdown if movement happens.

**Details on your device:**
 - Device: TS80P
 - Release: v2.16.D6AF465
 - Power adapter being used: included qc09
 - If this is an accelerometer related issue, please include its model number here: Where can i find this?

Acc: None
HW G 196
HW M 196
HW P 228

I've made some pictures of the PCB if necessary:
[image]

CvRXX, 2021-11-10:
I've also flashed bc7d106c9f19bb6a2132b196b2e3054323f0322c. The problem also occurred here!

Ralim, 2021-11-10:
Hello,

So this is the firmware indicating that when it probed for accelerometers, none responded on the communications bus.

Is this a newly flashed to ironos device or one that has been updated when this was first seen?

CvRXX, 2021-11-10:
Thanks for the fast response. It's newly flashed. I don't have the original firmware to test it on that.

Ralim, 2021-11-10:
Original firmware is over at https://minidso.com/

Would be good if you can confirm if it works on their firmware.
Most likely it is a hardware issue that would require an RMA; but its also possible its a new accelerometer model

CvRXX, 2021-11-10:
Flashed the original firmware. I've put it to defaults. Should the iron come back from sleep when I pick it up? Cause it doesn't. Maybe an RNA would be best then.

Ralim, 2021-11-10:
I believe it _should_.
I would be inclined to contact Miniware about an RMA / replacement at this point.

CvRXX, 2021-11-10:
I will do that!

CvRXX, 2021-11-10:
thank you for the help. I'll report back on what they say!

CvRXX, 2021-11-11:
Miniware asked for a video which I supplied. They agree on a broken accelerometer. They are going to send me a new soldering iron without tip. I've looked some more at the PCB photos I've shot and it seems that PB6 and PB7 are bridged. These pins seem to be responsible for I2C1. When the new soldering iron arrives I'm going to try to see if I can unbridge them. This would give me two soldering irons. 

Btw. I've told Miniware that I thought I could fix the iron but they still wanted to send me a new iron so credits to them!

CvRXX, 2021-11-11:
Issue was not software related.

closed 2021-11-11 by CvRXX

Ralim, 2021-11-11:
Glad you can get a replacement. Miniware are usually quite good with issues like this.

I don't think it would be a short as you would lose the OLED display of it was.

Could be a bad connection under the accelerometer though.

I hope replacement turns up soon for you :)

CvRXX, 2021-11-11:
Ah the display is also on the i2c. Yeah than it could be a bad accelerometer or a connection beneath it. That is a bit harder to fix :-p