Ralim/IronOS #1112: Xtar-PB2S PD3 battery-bank gets restarted after ~1s after getting into heating mode (TS80P)

Vinigas, 2021-11-11:
**Describe the bug**
PB2S batterybank is not able to keep up with TS80P powerdraw.

**To Reproduce**
1. Use XtarPB2S powerbank with 2 x 18650 batteries.
2. Use XtarPB2S included TypeC--USBA cable (it is capable 5V,9V,12V)
3. Connect it to TS80P and press heating.
4. After ~1s, power is cutted off. Guessing it causes by TS80P drawing more amperage, than PB2S is capable of.

**Expected behavior**
Hotend reached working temp.

**Details on your device:**
 
 - Device: TS80P
 - Release: v2.16.D6AF465 RC2
 - Power adapter being used: PB2S with it's included USBA--TypeC powercable

**Additional context**
Haven't tested yet on TS100, MHP30. Will keep this post updated as I test it out.
Then using TypeC--TypeC cable with this power bank, then it work without issue with TS80P heating stats: 9.1V 16.4W-max
It doesn't matter if battery is charged or not.

Ralim, 2021-11-11:
Hello,

So if in c-c mode the iron is only negotiating 9V, what voltage are you using with the A->C cable?

Sounds a lot like either the power bank can't source the power or the 18650's can't source the current.

Vinigas, 2021-11-11:
It negotiates 12V automatically.

Ralim, 2021-11-11:
When using the A->C cable it will be using QC; you may need to manually set it down to 9V for QC mode

Ralim, 2021-11-11:
When using PD the iron will select based on the advertised current rating of the charger so its "smarter" to avoid these issues.
For QC it _cant_ know the charger rating

Vinigas, 2021-11-11:
[image]

By specification, it should reach 12V1.5A (with setting max Watt draw to 18W in TS80P). Maybe it's battery type dependent.

Anyways, there is issue with powerbank going to sleep with default TS80P settings. Will try tweeking power-pulse setting.

Ralim, 2021-11-11:
@12V the TS80(P) needs 2.5A of current capacity.
This is why on PD its selecting 9V as its 1.5A at 9V

Vinigas, 2021-11-11:
[image]

Looks like it works stable with these tweaks (with USBA to TypeC cable):
o You need to enter into "Low-current output mode" as stated in PB2S instructions above, otherwise it shutdowns after fixed amount of time.
o On TS80P set max-wattage draw to 18W (this power-bank overcurrent protects at 19.5W)
o Set TS80P max QC voltage to 11V
o Set TS80P PD timeout = 1.

Then TS80P on cold-start shows: 9.9V 18.9W 
And then idle heat-ready shows: 10.9V 4.9W.

closed 2021-11-11 by Vinigas