Ralim/IronOS #1113: PD limited to 12V with Minix 65W charger

tve, 2021-11-13:
With my Minix Neo P1 66W charger a Pinecil running 2.16.RC2 fails to negotiate more than 12V using an Anker 65W cable (or an Anker 100W cable). This was also the case with previous releases, specifically V2.12 (I just upgraded from V2.12). The charger explicitly lists "20V 3.25A", which I believe should work, no?

 - Device: Pinecil
 - Release: 2.16 RC2
 - Power adapter being used: Minix Neo P1 66W, Anker 65W cable

How can I troubleshoot this?

Ralim, 2021-11-13:
20V 3.25A should work indeed. Though it does depend on if the power bank is advertising that correctly 😅

Does it have a label of all the supported voltages?

This code does not check the etag marker in cables, but the charger _may_. So far it hasn't been an issue though.

Do you have other devices that you connect via the usb-c power that work at the higher voltages ?

tve, 2021-11-13:
Sadly I don't have any other device other than a phone which doesn't show anything useful about PD. Here's a pic of the label:
[image]

The charger has 2x USB-C and 1x USB-A. It hadn't occurred to me before to try the USB-A with QC so I just did and it negotiates 19.6V. Interesting...

tve, 2021-11-13:
The Pinecil isn't actually usable with PD: it resets every few seconds. It's fine in sleep mode, which is what I had it in when I first posted this issue, but once asked to heat up it resets every 4-10 seconds. This did not use to happen with V2.12 where I used it successfully (on "11.8V") several times using this same changer and cable.
(At my desk I normally use 21V DC, the USB-C is only for mobility.)

Ralim, 2021-11-14:
If it's resetting when heating the most likely cause is the power adaptor. Once voltage is negotiated; PD bus is entirely silent normally. So it's not going to renegotiate and fail.

tve, 2021-11-14:
Strange thing though is that V2.12 also negotiated the same 12V PD setting and worked reliably, no crash. (Same adaptor and same cable.)

Ralim, 2021-11-14:
There has been a lot of tuning since then that has increased the maximum power we can achieve and tightened the regulation.

You might have progress is you use the power limiter feature to reduce the average power draw

tve, 2021-11-14:
Thanks for the info!

tve, 2021-11-19:
I just tried an Anker 65W PD / PPS charger and the negotiation doesn't work either: it also stops at 12V. Product link: https://us.anker.com/products/a2718
I only now saw the debug menu. It says it's doing QC, which explains the 12V. How do I force it to use PD?

Ralim, 2021-11-19:
You can't force PD, rather PD is the priority.
The iron won't start QC until PD has been times out (timeout setting in the menu) or rejected (power supple has refused to talk).

I would do a settings reset and check you don't get warnings that the PD chip can't be communicated with.
If comms to the chip is working then pd should work; as its basically a direct 1:1 connection to the connector

tve, 2021-11-19:
After a reset it indeed says no USB PD IC detected... So I have a defective iron and never realized 'cause it went to 12V due to QC and I normally use it with 21V DC input? Ughh...

Ralim, 2021-11-19:
The Pinecil is only rated for 20V operation due to the PD chip being present, though the PD part is "rated" to survive 21V; if your 21V is actually 24V then that may cause it.

So most likely is the PD IC (FUSB302) is damaged sadly :(

closed 2021-12-29 by Ralim