Ralim/IronOS #1120: Cannot build language other than EN

Devnol, 2021-11-20:
**Describe the bug**
Building with any language other than EN yields EN binaries

**To Reproduce**
1. Run make with `lang=(not EN) model=TS100`, for example `lang=FR`
2. Flash output hex to TS100
3. Boot iron
4. Menu is in EN

**Expected behavior**
Menu should be in selected language

**Details on your device:**
 
 - Device: TS100
 - Release: 2.16.BC7D106 (master)
 - Power adapter being used: N/A

alvinhochun, 2021-11-20:
Hm, the build system does not use a `lang` variable, instead you have to do `make model=TS100 firmware-FR`

Devnol, 2021-11-20:
Hmm, /Documentation/Development.md:12-20 says:
```
 Developing with command line tools & building a release
In the `source` folder there is a makefile that can be used to build the repository using command line tools.
when running the `make` command, specify which model of the device & the language you would like to use.

`make -j8 lang=EN model=TS80`

To build a release instead, run the build.sh script. This will update translations and also build every language for both TS100 and TS80 models.
```
I suppose that this has changed without updating the docs?

alvinhochun, 2021-11-20:
Yeah that doc is outdated... changed in https://github.com/Ralim/IronOS/pull/806

Devnol, 2021-11-20:
Oh cool, thanks

closed 2021-11-20 by Devnol