Ralim/IronOS #1124: Touch sleep mode

Vytatronikas, 2021-11-22:
Hello,
I understand this might not be the right place for my question, and my question may not be even related to the firmware. But I have a project where I would like to install the innards of the TS100 into a DIY soldering station and implement a different way of getting the iron go to sleep. The iron uses an accelerometer to trigger the sleep mode when it is not moving for a certain period of time. But would it be possible to implement the same sleep mode approach that is used in soldering stations? Namely, so that I can put the iron on a stand which connects the tip to a pin of micro, for example, pulling it to ground? Or, is this idea not possible with this iron?
Thank you very much for your excellent work.

Ralim, 2021-11-22:
So this logic is semi-there in that the Pinecil supports a hall effect sensor for sleep mode.

So if you used a pinecil for your handpiece, and fitted the hall sensor (its not fitted by factory) and then had a magnet in the base you would get this.

Otherwise it would be theoretically possible to set this up to happen with a GPIO on the microcontroller fairly easily, though not many of them are easy to access on the TS100
-----------------

Please note that if you are planning to install the control PCB somewhere else to the tip, you need to be very careful of what metals are used for that interconnection. As the copper metal spring contacts that mount to the PCB are part of the temperature measurement system as they are the "cold" junction for the thermocouple. Thus they need to be held at the same temperature as the temperature sensor on the PCB. You will need to be very careful on how you connect to the top and backto the PCB to maintain that the only dissimilar metal contact is at the PCB. Or move the copper contacts up to the tip (and therefore use copper for all wiring and other connectors) and move the cold junction temp sensor up there as well.

If you are thinking of just turning a Pinecil/TS100 into the handheld part then this doesnt apply.

River-b, 2022-02-12:
I installed a hall sensor on my Pinecils. and just use a neomydium Magnet on my stand and it instant goes to sleep whenever I'm in the stand.   that is easiest for DIY stand I think.  the sensor is only $2  and there is some info in this github somewhere on how to install it from other people asking about Hall sensor that you might be able to search for.   
      SI7210-B-00-IV  is the one on the Pinecil schematic at U14 and  the one I installed. 
It's not that hard to install the sensor but it's very small so that is a challenge.
        Ralim implemented hall debug setting also so you just follow debug number going up/down to figure out best placement for hall sensor. 
        Consider getting a Pinecil for your DIY stand instead of ts100 to use the very cool Hall Sensor feature which is perfect for STand sleep mode.  My pinecil goes to sleep within 2 seconds of touching the stand since the Magnet I put on the stand is sensed by the hall sensor.  And when I take it out, away from the magnet,  it immedicately heats up again.
 
I'm a member of pine64 discord channel if you want to find me but you could also join one of the many other Pine64 forums and someone in those might be around with experience with the hall sensor on Pinecil.  At first I installed the Hall sensor just for fun, but now that I have tested in in 5 different stands, it feels like a nice luxury feature of  high end Iron.