Ralim/IronOS #1135: TS80 unable to update firmware USB or DFU

zlice, 2021-12-06:
### Issue

Unable to update firmware on my TS80 through USB transfer or dfu-util.

Saw in another thread that a 'static'/dots screen may be a hardware issue, and my TS80 has done that for a long time. But even after the display shows `DFU: 3.45` there does not seem to be a way to update.

As you can see in dmesg, the serial number looks bogus.

Fdisk shows some ridiculous drive sizes.

And dfu-util says there's nothing there.

Guessing it's bad?

### Output

#### `dmesg` output

```
[20310.140086] sd 2:0:0:0: [sda] Attached SCSI removable disk
[20559.460769] usb 5-1.1: USB disconnect, device number 10
[20561.543149] usb 5-1.1: new full-speed USB device number 11 using xhci_hcd
[20561.656265] usb 5-1.1: New USB device found, idVendor=0483, idProduct=5720, bcdDevice= 2.00
[20561.656267] usb 5-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[20561.656269] usb 5-1.1: Product: STM32 Mass Storage
[20561.656270] usb 5-1.1: Manufacturer: STMicroelectronics
[20561.656270] usb 5-1.1: SerialNumber: 9ÿo\x06MU42\x17\x19\x04C
[20561.661274] usb-storage 5-1.1:1.0: USB Mass Storage device detected
[20561.661373] scsi host2: usb-storage 5-1.1:1.0
[20562.666489] scsi 2:0:0:0: Direct-Access     Virtual  DFU Disk              PQ: 0 ANSI: 2
[20562.666610] sd 2:0:0:0: Attached scsi generic sg0 type 0
[20562.666827] sd 2:0:0:0: [sda] 4096 512-byte logical blocks: (2.10 MB/2.00 MiB)
[20562.666962] sd 2:0:0:0: [sda] Write Protect is off
[20562.666963] sd 2:0:0:0: [sda] Mode Sense: 03 00 00 00
[20562.667092] sd 2:0:0:0: [sda] No Caching mode page found
[20562.667095] sd 2:0:0:0: [sda] Assuming drive cache: write through
[20562.683161]  sda:
[20562.684060] sd 2:0:0:0: [sda] Attached SCSI removable disk
```

#### `fdisk` output

```
fdisk -l /dev/sda
Disk /dev/sda: 2 MiB, 2097152 bytes, 4096 sectors
Disk model: DFU Disk        
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x6f20736b

Device     Boot      Start        End    Sectors   Size Id Type
/dev/sda1        778135908 1919645538 1141509631 544.3G 72 unknown
/dev/sda2        168689522 2104717761 1936028240 923.2G 65 Novell Netware 386
/dev/sda3       1869881465 3805909656 1936028192 923.2G 79 unknown
/dev/sda4                0 3637226495 3637226496   1.7T  d unknown

Partition table entries are not in disk order.
```

#### `dfu-util` output

```
dfu-util -D TS80_EN.bin 
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
dfu-util: No DFU capable USB device available

```

Ralim, 2021-12-09:
Hmmm,

So; dfu-util won't work with a TS80, but those disk details definitely look bogus.

The serial number is usually bogus iirc but it should just present a 2MB FAT32 partition not gibberish.

What where you doing before these issues ?

zlice, 2021-12-09:
I can't remember the last time I updated it, so I really don't know.

Ralim, 2021-12-19:
So fdisk I think falls over on these due to the way Miniware is faking things.
The serial number being gibberish is normal, I believe they are dumping the IC serial in there as raw bytes not ascii.

For reference this is my TS80P
```
[  +3.318780] usb 5-1: new full-speed USB device number 55 using xhci_hcd
[  +0.165489] usb 5-1: New USB device found, idVendor=0483, idProduct=5720, bcdDevice= 2.00
[  +0.000003] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000001] usb 5-1: Product: STM32 Mass Storage
[  +0.000001] usb 5-1: Manufacturer: STMicroelectronics
[  +0.000001] usb 5-1: SerialNumber: Wÿl\x06rqRR(5\x01g
[  +0.005016] usb-storage 5-1:1.0: USB Mass Storage device detected
[  +0.000142] scsi host5: usb-storage 5-1:1.0
[  +1.028071] scsi 5:0:0:0: Direct-Access     Virtual  DFU Disk              PQ: 0 ANSI: 2
[  +0.000398] sd 5:0:0:0: [sdc] 4096 512-byte logical blocks: (2.10 MB/2.00 MiB)
[  +0.000114] sd 5:0:0:0: [sdc] Write Protect is off
[  +0.000002] sd 5:0:0:0: [sdc] Mode Sense: 03 00 00 00
[  +0.000109] sd 5:0:0:0: [sdc] No Caching mode page found
[  +0.000001] sd 5:0:0:0: [sdc] Assuming drive cache: write through
[  +0.005572]  sdc:
[  +0.000610] sd 5:0:0:0: [sdc] Attached SCSI removable disk
```

zlice, 2021-12-20:
Well it died. Idk if getting a ts80p was a mistake or not but that's what I did...closing

closed 2021-12-20 by zlice