Ralim/IronOS #1148: Pinecil GD32VF103 Bootloader reset

slackista, 2021-12-28:
`dfu-util -l` causing Pinecil bootloader to crash.

**To Reproduce**
1. Hold `-` in Pinecil.
2. Connect to usb.
3. run `dfu-util -l`
4. Expect for kernel usb errors
5. Unplug and start over.

One of ~20 plugs `dfu-util` works fine and I am able to `-l` or even flash firmware with `-D`

- Pinecil with SC7A20
- dfu-util 0.11
- libusb 1.0.24

Checked with 3 different usb-c cables. With every socket in PC (type C and type A)

Logs after plug in:
```
kernel: usb 1-1: new full-speed USB device number 81 using xhci_hcd
kernel: usb 1-1: New USB device found, idVendor=28e9, idProduct=0189, bcdDevice= 1.00
kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
kernel: usb 1-1: SerialNumber: 吳䩂
```

Logs after `dfu-util -l`:
```
kernel: usb 1-1: reset full-speed USB device number 81 using xhci_hcd
kernel: usb 1-1: device descriptor read/64, error -71
kernel: usb 1-1: device descriptor read/64, error -71
kernel: usb 1-1: reset full-speed USB device number 81 using xhci_hcd
kernel: usb 1-1: device descriptor read/64, error -71
kernel: usb 1-1: device descriptor read/64, error -71
kernel: usb 1-1: reset full-speed USB device number 81 using xhci_hcd
kernel: usb 1-1: Device not responding to setup address.
kernel: usb 1-1: Device not responding to setup address.
kernel: usb 1-1: device not accepting address 81, error -71
kernel: usb 1-1: reset full-speed USB device number 81 using xhci_hcd
kernel: usb 1-1: Device not responding to setup address.
kernel: usb 1-1: Device not responding to setup address.
kernel: usb 1-1: device not accepting address 81, error -71
kernel: usb 1-1: USB disconnect, device number 81
kernel: usb 1-1: new full-speed USB device number 82 using xhci_hcd
kernel: usb 1-1: device descriptor read/64, error -71
kernel: usb 1-1: device descriptor read/64, error -71
kernel: usb 1-1: new full-speed USB device number 83 using xhci_hcd
kernel: usb 1-1: device descriptor read/64, error -71
kernel: usb 1-1: device descriptor read/64, error -71
kernel: usb usb1-port1: attempt power cycle
kernel: usb 1-1: new full-speed USB device number 84 using xhci_hcd
kernel: usb 1-1: Device not responding to setup address.
kernel: usb 1-1: Device not responding to setup address.
kernel: usb 1-1: device not accepting address 84, error -71
kernel: usb 1-1: new full-speed USB device number 85 using xhci_hcd
kernel: usb 1-1: Device not responding to setup address.
kernel: usb 1-1: Device not responding to setup address.
kernel: usb 1-1: device not accepting address 85, error -71
kernel: usb usb1-port1: unable to enumerate USB device
```

Ralim, 2021-12-28:
I have seen this with some USB controllers or poor quality USB cables.

Please try other USB ports and other usb-c cables.

Good quality USB2.0 ones tend to be the best ones for me.

slackista, 2021-12-28:
Turns out all 3 cables are good. The issue is usb autosuspend. Turn **off** Linux usb autosuspend fixed this wired issue. Closing.

closed 2021-12-28 by slackista

edubxb, 2022-01-19:
There's [gist](https://gist.github.com/edubxb/f39d41da4c3b57c481b9f4fe9c73d206) with the udev rules I use to successfully flash the pinecil.