Ralim/IronOS #1150: QC voltages above 5V not supported on RC2

cybernesto, 2021-12-29:
**Describe the bug**
Using the stable release v2.15 I was able to use QC voltages up to 12V using the USB 3.0 outputs of my RP-PC136 charger. The same charger provides 65W on the USB-C outputs and this works with 20V using a USB-C to USB-C cable flawlessly. The latest RC2 only allows 5V to be used in QC mode, so no soldering is possible using this cable anymore.

Even though USB PD is my preferred solution on this charger, I would like to keep QC as an option in case I get a portable charger or another power option. Currently it would mean reverting to v2.15.

**To Reproduce**
1. Connect an USB-3.0 to USB-C cable to the QC output of the RP-PC136 charger
2. Check the available voltage.

**Expected behavior**
At least 12V should be available, as they were in the stable release. 

**Details on your device:**
 - Device: Pinecil
 - Release: 2.16.D6AF465
 - Power adapter being used: RAVPower PD Pioneer 65W 4-Port Desktop Charger RP-PC136

**Additional context**
Different cables made no difference.

Ralim, 2021-12-29:
To test why this is not kicking in, could you give a try of setting the PD timeout to like 10 (1 second) 

Does the charger bear the Qualcomm QC logo ?

cybernesto, 2021-12-29:
Indeed! Setting the PD timeout to 10 enables the 12V again. The charger does not have the Qualcomm Logo. The QC port on the back is called iSmart Output and is rated as 5V 3A, 9V 2A, 12V 1.5A, 18W Max.

cybernesto, 2021-12-29:
Setting the PD timeout to 1 also works btw.

discip, 2021-12-29:
@cybernesto
Please don't forget to close this issue, if your case is solved. 😊

thank you

cybernesto, 2021-12-29:
PD timeout was not required for v2.15 but is now required for v2.16 when using an RP-PC136 charger.

closed 2021-12-29 by cybernesto