Ralim/IronOS #1154: Automatic input Voltage detection

RoadyFPV, 2022-01-02:
It would be nice if the soldering iron recognizes the connected lipo and automatically sets the correct cut-off voltage. So that you do not have to manually specify the used Lipo 3S, 4S, 5S, 6S.

This would be very useful if you often use different LIPOs

Johnex, 2022-01-02:
The issue with this is that cell voltages overlap. A run down 5S lipo has the same voltage as a fully topped off 4S. How would you know if it's a 4S or 5S in that case? If you are not paying attention, and the wrong cell type is detected, then your battery will go bellow a safe voltage. This is the reason there isn't automatic detection already.

RoadyFPV, 2022-01-03:
Ok, What Do you think about an Option to select the cells on Boot up?

Ralim, 2022-01-03:
This has been discussed at least twice in the past.

Every time it doesn't get anywhere as you can't confidently automatically select the cell count based on voltage alone. 

Sensing at boot up doesn't help since you dont know if the pack is charged or not.

Sensing based on voltage droop would work, except that people often use poor wiring which would falsely raise the alert as well. Though it has more merit then just guessing. It would also be a lot for work to tune such a system.

RoadyFPV, 2022-01-03:
Yes, ok I understood the problem of the automatic cell recognition.

How would it be to have a selection option under energy options. for example "on Startup".

Then after plugging in, the soldering iron asks which battery you are using, and you can switch between the options 3S,4S,5S,6S with + and -.

Ralim, 2022-01-03:
It would be possible, though not a trivial amount of work to add.

If someone wants to code it into a PR, more than happy to look at it and most likely merge it in. I wont have time to add it for a bit though at the moment :/