Ralim/IronOS #1156: Cannot flash latest 2.16 into my TS100 (board V1)

AndroidGX, 2022-01-03:
Hi there,

Coming from last 2.15 official, I tried to flash 2.16 but I have the following error when I try to slide the usual FR hex file into my TS100 reader : https://i.ibb.co/855NkBQ/explorer-fw-Gv-PIP4a-I.png
I unplugged the USB while still in DFU 3.45 but the reader is no more here.
Since then my iron is now no more working at all... :/ 
I tried to flash back 2.15 with the same issue.
Tried to change USB cable with same issue.
Rebooting my computer with same issue.

How can I recover my TS100 now ? damn
Many thanks in advance for your help

AndroidGX, 2022-01-03:
Here is a quick video to show the issue : 
https://user-images.githubusercontent.com/5033546/147912301-e496bca0-01e6-4779-83c0-358af4128c86.mp4

Ralim, 2022-01-03:
This is a computer related issue, not this firmware sadly.
Usually this is anti-virus software causing grief.

Things to try:
1. Different usb cable (unlikely)
2. Different USB port
3. Different computer
4. Double copy; after the error the device should re-appear, and try it again without powering the iron off
Make sure you are extracting the files from the zip first.

hugomon, 2022-01-03:
I have the same problem, with the difference that mine is not bricked. I have tried several quality cables, ports and computers, the problem is not there. If you try to reload the firmware after the error, it simply shows that the file is too large for the file system.

Ralim, 2022-01-03:
I tested the _EN files to sanity check one all of my units where without issues.
So I _believe_ the files are good.

Issues with copying to the Miniware irons are almost always to do with the host system or operating system sadly.
Especially ones to do with it disconnecting mid transfer, as these are to do with how their bootloader fakes the usb drive to the computer.

hugomon, 2022-01-03:
I've already found the error, I was trying to load the dfu file instead of the hex, I hope AndroidGX's problem was the same.

AndroidGX, 2022-01-03:
Thanks for your answers,
I just tried in mom computer on Windows 7 (instead of Windows 10), and the iron wasn't recognized properly in the USB process.
After several tries with gently pushing the USB into the connector (not till 100%) the iron was finally recognized and pushed the hex file, and this time worked properly.
Glad to find back in a working state my TS100, was a bit worried to buy another one just due to this update.
Many thanks,
Cheers'
👍

closed 2022-01-03 by AndroidGX

volt-k, 2022-01-23:
Hi, I just had the same issue with my TS100. Booting Windows 10 into "Safe Mode" helped, firmware update worked straight away. Glad I've found this thread. I think it's worth metnioning it on wiki.