Ralim/IronOS #1157: feature request: default to automatic screen rotation when an accelerometer is available

evils, 2022-01-03:
**Additional context**
i'm left-handed and annoyed every time i flash a new release

**Additional note**
just to make sure it's mentioned in an issue, the "no tip" icon in 2.16 always points left, so points the wrong way when the screen is rotated left

discip, 2022-01-04:
Good evening.

First of all:
> i'm left-handed and ***annoyed every time*** . . .

## This is NOT the way to open an issue!

- If you don't like it, than just stick with the proprietary firmware.

- You should be ***grateful***, that you are allowed to get this firmware ***for nothing***.

The way you expressed yourself, gives the impression, that the developer owes you to design this firmware solely according to your wishes.

But . . .

May be it was not your intention and you did not think about the fact, that this could be perceived the wrong way.

There is a saying, that goes like this: ***The tone makes the music.***
Please take this into account.

Maybe you should first express your appreciation, for the work, that was put into this and tell the developer how much better ***his firmware*** is, compared to the stock one.

One more saying:
*We reap what we sow.*

So, maybe your issue would be addressed by now already.

> the "no tip" icon in 2.16 always points left

That's not *entirely* true, since this appears only, if the ***detailed idle screen*** is chosen.

By the way: This behavior was present even before the replacement of the according text.

kind regards

evils, 2022-01-05:
> This is NOT the way to open an issue!

you're right, the lighthearted tone i intended does not come across in the comment
it is not helped by me removing all the preamble to "additional context"
(which i removed because i found the title an adequate description)
and i failed to take into account this would now be the first message
(i may have cleaned up the rest of the template after writing that)

i also did not foresee that the issue would be immediately assigned to someone,
which i feel makes my comment read as targeted

>     * You should be _**grateful**_, that you are allowed to get this firmware _**for nothing**_.

i am grateful, and believe that in return for the privilege of using this software, i am obliged to report bugs and suggest ways to improve it

> May be it was not your intention and you did not think about the fact, that this could be perceived the wrong way.
> 
> There is a saying, that goes like this: _**The tone makes the music.**_ Please take this into account.

do note, your tone prior to these lines, is not one of deescalation

> Maybe you should first express your appreciation, for the work, that was put into this and tell the developer how much better _**his firmware**_ is, compared to the stock one.

i'm in the very privileged position of never having used the TS100 stock firmware, as i'm using a pinecil

sandmanRO, 2022-01-05:
Hello @discip, 

Before anything Happy New Year!

Regarding that rotation issue of the missing tip icon, you already beautifully implemented it for the graphic layout mode so maybe it's worth doing the same thing for detailed layout mode, that is copying the following block of code (starts at line 936 with GUIThread.cpp):

#ifdef OLED_FLIP
          if (!OLED::getRotation()) {
#else
          if (OLED::getRotation()) {
#endif
            // in right handed mode we want to draw over the first part
            OLED::drawArea(55, 0, 41, 16, disconnectedTipIconFlip);

          } else {
            OLED::drawArea(0, 0, 41, 16, disconnectedTipIcon);
          }

and pasting it over (replacing) the line 916 of the same file (the line that needs to be replaced with the code above is this one
OLED::drawArea(0, 0, 41, 16, disconnectedTipIcon);). Just a thought...

discip, 2022-01-05:
Good evening @evils,

Thank you for clearing this up. 😊
> do note, your tone prior to these lines, is not one of deescalation

This was not intended to attack you, it was rather *slamming onto the brakes*.
I read this passage (which you refer to) several times, prior to posting and my goal was to stay objective.
If I wanted to be rude, there would be no turning-point *(But . . . )*!

In general, I do not like confronting people on my behalf, but in this case it was not about me.

So, this should be settled now. 🤝

discip, 2022-01-05:
@sandmanRO
Good evening Sir,
thank you for telling me what to do.
Still I don't quite get what exactly to copy:
Should it be this part?

<details>
<summary>
snippet (936 - 993) to be copied
</summary>

```C#
#ifdef OLED_FLIP
      if (!OLED::getRotation()) {
#else
      if (OLED::getRotation()) {
#endif
        OLED::drawArea(12, 0, 84, 16, idleScreenBG);
        OLED::setCursor(0, 0);
        gui_drawBatteryIcon();
      } else {
        OLED::drawArea(0, 0, 84, 16, idleScreenBGF); // Needs to be flipped so button ends up
                                                     // on right side of screen
        OLED::setCursor(84, 0);
        gui_drawBatteryIcon();
      }
      tipDisconnectedDisplay = false;
      if (tipTemp > 55)
        tempOnDisplay = true;
      else if (tipTemp < 45)
        tempOnDisplay = false;
      if (isTipDisconnected()) {
        tempOnDisplay          = false;
        tipDisconnectedDisplay = true;
      }
      if (tempOnDisplay || tipDisconnectedDisplay) {
        // draw temp over the start soldering button
        // Location changes on screen rotation
#ifdef OLED_FLIP
        if (!OLED::getRotation()) {
#else
        if (OLED::getRotation()) {
#endif
          // in right handed mode we want to draw over the first part
          OLED::fillArea(55, 0, 41, 16, 0); // clear the area for the temp
          OLED::setCursor(56, 0);

        } else {
          OLED::fillArea(0, 0, 41, 16, 0); // clear the area
          OLED::setCursor(0, 0);
        }
        // If we have a tip connected draw the temp, if not we leave it blank
        if (!tipDisconnectedDisplay) {
          // draw in the temp
          if (!(getSettingValue(SettingsOptions::CoolingTempBlink) && (xTaskGetTickCount() % 260 < 160)))
            gui_drawTipTemp(false, FontStyle::LARGE); // draw in the temp
        } else {
          // Draw in missing tip symbol

#ifdef OLED_FLIP
          if (!OLED::getRotation()) {
#else
          if (OLED::getRotation()) {
#endif
            // in right handed mode we want to draw over the first part
            OLED::drawArea(55, 0, 41, 16, disconnectedTipIconFlip);

          } else {
            OLED::drawArea(0, 0, 41, 16, disconnectedTipIcon);
          }
```

</details>

snippet (916) to be replaced

```C#
        OLED::drawArea(0, 0, 41, 16, disconnectedTipIcon);
```

sandmanRO, 2022-01-05:
Hello @discip, I so apologize for the typo....the start copy line is 983 (and not 936 as I wrongfully state above)... precisely only the code below shall be copied (then pasted over / instead of line 916):

#ifdef OLED_FLIP
          if (!OLED::getRotation()) {
#else
          if (OLED::getRotation()) {
#endif
            // in right handed mode we want to draw over the first part
            OLED::drawArea(55, 0, 41, 16, disconnectedTipIconFlip);
          } else {
            OLED::drawArea(0, 0, 41, 16, disconnectedTipIcon);
          }

discip, 2022-01-05:
@sandmanRO
Thank you for the instant response!
You have absolutely no reason to apologize, since I am the one who need to get help!

kind regards

discip, 2022-01-06:
@sandmanRO
Thank you for having pushed me in the right direction! 👍

@evils
I hope you like the new detailed lefty-mode. 😊
https://github.com/Ralim/IronOS/pull/1165

evils, 2022-01-06:
i like it a lot, it has the "good UI is invisible" feel to it

discip, 2022-01-06:
Glad, you like it. 😊

discip, 2022-01-08:
@evils
Since everything you requested was merged, I think it is time to close this.

evils, 2022-01-08:
oh woeps, i totally missed #1168, thanks!
@discip you may want to try [github's automatic issue closing feature](https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/using-keywords-in-issues-and-pull-requests#linking-a-pull-request-to-an-issue) next time

closed 2022-01-08 by evils

discip, 2022-01-08:
This is new to me, thanks for the hint. 😃