Ralim/IronOS #1159: settings sometimes aren't saved

evils, 2022-01-03:
**Describe the bug**
several times (mostly after flashing a new firmware release) i've gone through the settings menu to configure my preferences
and later found these settings were not saved

i suspect i unplugged the power source to exit the menu rather than cycle through it to get back to the main screen

**To Reproduce**
1. reset settings
2. set display orientation to automatic and continue to the next menu item
3. rotate pinecil to verify the setting has taken effect
4. unplug the power source and plug it back in
5. see that the screen setting did not take effect

**Expected behavior**
settings which take immediate effect, to stay in effect after a power cycling

**Details on your device:**
 
 - Device: Pinecil
 - Release: [2.15, 2.16]
 - Power adapter being used:[desktop USB port, 100W brick]

Ralim, 2022-01-03:
Settings are only saved when exiting the menu by design to reduce wear on the flash :/

Child maybe save on exiting the menu groups perhaps?

evils, 2022-01-03:
hmm, i guess display rotation is one of the few settings that's actually noticeable in the settings menu
i suppose not having the settings take effect until they're actually saved would be worse UX because of that
having an indication of "settings saved" at the end of the menu could get annoying, but would have explained the behaviour

saving at the end of each modified settings group/submenu could be a fair middle ground
though given the choice i'd probably go for the one that incurs the least wear

mentioning that the settings are only saved at the end of the menu in the documentation may help
but honestly, i did not check that before opening this issue

and i just checked, it is mentioned [here](https://github.com/Ralim/IronOS/blob/153e7ba36f8a79c2fcfc28fe8d3398224176dce8/Documentation/Menu.md#settings-menu)

closed 2022-01-03 by evils