Ralim/IronOS #1160: with "temp change long: 20" lowering temperature rolls over to 450°C

evils, 2022-01-03:
**Describe the bug**
as stated in the title, it does not roll over when raising the set temperature

**To Reproduce**
1. reset settings
2. set "temp change long" to 20
3. proceed through the menu to ensure the setting is saved (#1159)
4. enter temperature change mode, lower temperature by holding `-`
5. see the temperature go from 10°C to 450°C
6. optionally try raising the temperature and see that the issue doesn't occur in that direction

**Expected behavior**
the set temperature change to stop at 10°C regardless of the size of the steps or the speed of change

**Details on your device:**
 
 - Device: Pinecil
 - Release: 2.16
 - Power adapter being used: 100W brick

discip, 2022-01-06:
Good afternoon @evils,
if that https://github.com/Ralim/IronOS/pull/1164#issuecomment-1006298660 is the case, that please close this issue. 😊

thank you
kind regards

evils, 2022-01-06:
> please close this issue.

i will when it's merged :)

closed 2022-01-06 by evils