Ralim/IronOS #1170: TS100

mimimmim, 2022-01-07:
**Describe the bug**
After apdate last version display blinkss and shows pieces of images

**To Reproduce**
[image]

**Expected behavior**
Tried other version - blinkss and shows pieces of images.
Version 2.0 - blinking stops, but it freezes in 3 seconds, in ~30 seconds goes to mainscreen, and again freezes.  

**Details of your device:**
 - Device: TS100
 - Release: unknown, DFU 3.47
 - Power adapter being used: 19v, 4.74a

**Additional context**
May be the problem is version of DFU, how can I downgrade DFU?

Ralim, 2022-01-07:
Hello,

This could be a hardware issue,but I'm not sure yet.

To confirm, does it always do this at startup? As it looks like you have changed to detailed mode, could you try doing  a factory reset in the advanced menu.

Does the screen do anything weird in dfu mode?

Ralim, 2022-01-07:
Ah good point.

I would still try a settings reset if possible.

If that doesn't help it any be worth checking with the miniware firmware. But suspect i2c has an issue

VaZso, 2022-01-08:
Do I understand well it seems the display looks correct using firmware version 2.0?

Settings reset and trying with miniware frimware to see if hardware is working well a really good starting point.
If it works, it would be good to know if you have tried any version of IronOS earlier which has completely worked (and you may try that version now).

It is not really clear for me how V2.0 works for you, so what you mean about freezing - I suspect the screen is not changing but it goes to main screen later, so firmware itself is still running.

Anyway, does soldering work (maintains temperature) despite of screen problems?

It would be good if you could answer these questions above.

Anyway, I have used similar OLED displays in my project(s) earlier also over I2C and SPI which includes the very same display what TS100 / Pinecil use and also bigger 128x64 pixel versions.
I had to modify the code which was initially worked well for me when later order of displays had same change of its display controller which has caused display distortions in newer displays.
Finally I could modify its driver to work correctly using also older and newer displays.

It does not mean it is the problem with your TS100 but it is also a possible option if stock firmware is working and you have not used IronOS earlier (so answering questions above may help)...

mimimmim, 2022-01-08:
I hope on video there are all answers, if not then make me know what needs alse.

Ralim 2.16
https://youtu.be/aa6OgD87HzE
https://youtu.be/Z59ViBw4ut4

Ralim 2.0
https://youtu.be/-AqxpnkaB8E
https://youtu.be/_PYzawMOnoo

DFU
https://youtu.be/3o4zjSsKq-Y

Stock 2.18/2.17/2.12/2.11
https://youtu.be/so8s6mx1VDo

mimimmim, 2022-01-08:
"As it looks like you have changed to detailed mode, could you try doing a factory reset in the advanced menu."

I can try if you tel me how.

VaZso, 2022-01-08:
Your videos seem to be "private"...

mimimmim, 2022-01-08:
https://cloclo-stock6.datacloudmail.ru/stock/get/6AQ37SEqs8L6gXvCxzNcuXkJjWdSey9CmYghmZaRxfutKgoM5oKgW2CviJPE54SfGMh6nb75b734/video%20TS100.zip?x-email=mimim%40bk.ru

zip video.

mimimmim, 2022-01-08:
youtube is processing video

mimimmim, 2022-01-09:
Any ideas?

mimimmim, 2022-01-11:
display works only on version 2.0 2.1 and 2.2
How erase logo image?

Gianluxx, 2022-01-11:
Hey i am having similiar problems. Bought the ts100 from China and arrived today. Link: https://a.aliexpress.com/_uxYTkh
Works perfect with official firmware (arrived with 2.17, now upgraded to 2.20).
With ironOs 2.16 the screen is black and nothing works.
With ironOs 2.15 the screen glitches but i can "see" the menus working but glitched.
The DFU works without glitches.
Every flash gives .RDY back. Double flash gives .ERR on the second one.

mimimmim, 2022-01-11:
Gianluxx, what version DFU on you device?

VaZso, 2022-01-11:
Anyway, when have you purchased your TS100?

It is not an easy task to find out a problem without a device which has problems.
It may be related to I2C or if these are relatively new devices, I would think it may happen their display controller has been changed (I mean compared to earlier TS100 or what Pinecil uses).

If I remember well, originally these displays had SSD1306 display controller and I think I had to change my code because of SSD1307 or SSD1308 version has appeared and it seems there is also SSD1309 version exists (so that is the display what installed in TS100).

It may be a starting point to compare their data sheets and / or check if they have a register to check which version of display controller it has (and how to show it). :)

...maybe it also worths a try to simply slowing down I2C clock.

Gianluxx, 2022-01-11:
> Gianluxx, what version DFU on you device?

3.47

Gianluxx, 2022-01-11:
> Anyway, when have you purchased your TS100?

8 days ago

Ralim, 2022-01-12:
We are already running a slow 75kHz I2C clock for the Miniware units due to some units having missing I2C pullup resistors, so I doubt that slowing it down will help with this.

You are running a much newer TS100 bootloader than historically we have seen.
Could you open the back of the unit and take a photo of the small little board that is attached, want to check if you have received a unit that uses an official STM32F103 part or something else.

The fact that it doesnt work on older miniware code either makes me suspect we have another part swap that has gone on.

Gianluxx, 2022-01-12:
[image]

Ralim, 2022-01-12:
Ah, that would explain the issue.
I was told they were not swapping out the GD32 _yet_.
I have one of that model on the way to get working.

Gianluxx, 2022-01-12:
Perfect thank you. We will wait

Ralim, 2022-02-05:
Going to close this one in favour of #1177

closed 2022-02-05 by Ralim