Ralim/IronOS #1173: not able to flash logo.hex | img2ts100.py not working (Win10)

discip, 2022-01-09:
**Describe the bug**
1. Not able to flash any *logo.hex* file to the Iron.
2. Not able to flash *_clear_logo.hex*

resulting in ***.ERR***

3. *img2ts100.py* not working

**To Reproduce**
follow **[Startup Logos](https://github.com/Ralim/IronOS/blob/master/Documentation/Logo.md)**

get this:
```py
    <!DOCTYPE html>
    ^
SyntaxError: invalid syntax
```

**Expected behavior**
successfully convert png to hex

**Details of your device:**
**TS80P** running latest build https://github.com/Ralim/IronOS/commit/5058e18138635b87978e24db58bf4c58c60e0a15

**Additional context**
I did:
```py
python3 -m pip install Pillow
```

Ralim, 2022-01-09:
You have not downloaded the python code correctly, from that error I suspect you have saved the github webpage preview not the file itself. Need to hit the `raw` button to get to this page https://raw.githubusercontent.com/Ralim/IronOS/master/Bootup%20Logo/python_logo_converter/img2ts100.py

Then save and run that code

discip, 2022-01-09:
Ok! I see!🤦‍♂️
Will give that a shot and report back.

discip, 2022-01-09:
That indeed worked, but I am not able to remove the logo again using *_clear_logo.hex*.

Ralim, 2022-01-09:
Did you do the double copy trick for the hex file?

discip, 2022-01-09:
Yes! I did.

Ralim, 2022-01-09:
>ps: I thought the custom logo would replace the quick flashing miniware logo, but that did not happen.
Is that not replaceable?

That is part of their bootloader, so cant remove it without removing their bootloader.

discip, 2022-01-09:
> That is part of their bootloader, so cant remove it without removing their bootloader.

Ok. I see. So someone should rewrite the bootloader to solve this right? 😁

Ralim, 2022-01-09:
So you can flash the DFU bootloader instead, which fixes this, but also removes the usb-drive functionality entirely and you have to use dfu-util to flash instead.

Thats about all you can do to remove their logo for now. I have not undertaken writing a "new" bootloader that does the usb stuff. I have been thinking about writing a usb-dfu one that I maintain for these units. But low priority project for now.

Just dont have enough time to build all the things... :/

discip, 2022-01-10:
No problem! 👍
I am good with that. If there will be one day a *usb-dfu one*, I will be happy, but until than I will just stick to the current situation. 😊

Could it be that *_clear_logo.hex* does not work due to my iron using **DFU3.50** instead **DFU3.45**?

discip, 2022-01-10:
> So you can flash the DFU bootloader instead, which fixes this, but also removes the usb-drive functionality entirely and you have to use dfu-util to flash instead.

Is there a way to go back to the *miniware* bootloader after that?

discip, 2022-01-12:
Am I the only one to not get this to work?
# Could someone please check, if `_clear_logo.hex` is working?

thanks in advance

discip, 2022-01-19:
Anyone?

Ralim, 2022-01-20:
Planning to test next time I boot windows..it's been a bit

L1cardo, 2022-02-14:
Same here, I can not remove the custom boot logo either

L1cardo, 2022-02-15:
> If you are on Pinecil or a dfu bootloader you can just erase the flash easily.
> Otherwise could you give this a test:
> [erase_test.zip](https://github.com/Ralim/IronOS/files/8060244/erase_test.zip)

I am using TS80P, and this one works!

PR #1232 (2022-03-13): Logo rework

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [ ] There are no breaking changes

* **What kind of change does this PR introduce?**

Support for new logo support that (a) fixes the old endian issue and (b) allows animation.
Removes logo's from this repo and instead places them in the Meta repo.

* **What is the current behavior?**

* **What is the new behavior (if this is a feature change)?**

* **Other information**:

Closes #1173
Closes #1228
Closes #617
Closes #446
Inches closer to supporting #292
closed 2022-02-15 by discip