Ralim/IronOS #1177: v2.16 does not work on TS100 with DFU 3.48

DimaStebaev, 2022-01-11:
**The bug description**
After flashing v2.16 on TS100 with DFU 3.48 the iron becomes a brick.

**To Reproduce**
1. Take brand new TS100 with DFU 3.48
2. Download TS100.zip file from [the v2.16 release](https://github.com/Ralim/IronOS/releases/tag/v2.16)
3. Unzip TS100.zip
4. Press button A and connect TS100 to windows PC
5. Copy TS100_RU.hex to TS100

**Expected behavior**
After connecting the power, the iron should turn on and operate with IronOS.
Instead it continues to be turned of. Nothing happens.

**Details of your device:**
 - Device: TS100
 - Release: Can't get hardware release, late 2021 model, DFU 3.48
 - Power adapter being used: YH24-3

**Additional context**
After flashing the iron still enters DFU mode.
Default firmware v2.18 does not work as well and causes the same behavior.
Default firmware v2.20 works correctly.

Ralim, 2022-01-12:
Please see my comment at the end here : https://github.com/Ralim/IronOS/issues/1170

I suspect that a part swap has gone on and IronOS does not support your hardware. Could you confirm some photos of the inside of your unit?

PUF52, 2022-01-13:
> **Описание бага** После прошивки v2.16 на TS100 с DFU 3.48 железо становится кирпичом.
> 
> **Воспроизвести**
> 
> 1. Возьмите новенький TS100 с DFU 3.48.
> 2. Загрузите файл TS100.zip из [версии v2.16.](https://github.com/Ralim/IronOS/releases/tag/v2.16)
> 3. Разархивируйте TS100.zip
> 4. Нажмите кнопку A и подключите TS100 к ПК с Windows.
> 5. Скопируйте TS100_RU.hex в TS100
> 
> **Ожидаемое поведение** После подключения питания утюг должен включиться и работать с IronOS. Вместо этого он продолжает быть выключенным. Ничего не произошло.
> 
> **Детали вашего устройства:**
> 
> * Устройство: ТС100
> * Релиз: не удается получить аппаратную версию, модель конца 2021 г., DFU 3.48.
> * Используемый адаптер питания: YH24-3
> 
> **Дополнительный контекст** После прошивки железо все равно входит в режим DFU. Прошивка по умолчанию v2.18 не работает и вызывает такое же поведение. Прошивка по умолчанию v2.20 работает корректно.

Дима, здравствуй. Завтра получу ts 100, но после твоего поста вкрались сомнения, можно ли его прошивать на версию v2.16. Не получу ли кирпич? Спасибо. У вас проблема решилась? Спасибо.

paulfertser, 2022-01-13:
On Thu, Jan 13, 2022 at 02:35:38AM -0800, PUF52 wrote:
> Дима, здравствуй. Завтра получу ts 100, но после твоего поста вкрались сомнения,
> можно ли его прошивать на версию v2.16. Не получу ли кирпич? Спасибо. У вас
> проблема решилась? Спасибо.

Folks, what's happening? How come the messages in Russian end up
appearing in the issue tracking system? If you're just writing them
that way, pretty please consider using English instead so that they
would be easy to understand for everybody, not just the
Russian-speaking minority.

Regarding potential "bricking", the vendor bootloader never overwrites
itself so it should always remain accessible. But even if something
happens to it one can always attach an SWD adapter via the USB
connector and reflash. I'm using a real DFU bootloader with dfu-util
and it's way more comfortable and reliable compared to the vendor's
mass storage silliness.

-- 
Be free, use free (http://www.gnu.org/philosophy/free-sw.html) software!
***@***.***

DimaStebaev, 2022-01-13:
Hi @PUF52.
I suspect that the manufacturer has changed a hardware and v2.16 needs to be modified to work properly on new revision.
As @paulfertser  said there should be pretty low risk to completely brick a device with default DFU. Another words you may try and roll back in case of failure. I suggest you to write down somewhere a version of default firmware your TS100 will have right after you receive it.

PUF52, 2022-01-13:
Спасибо

discip, 2022-01-13:
@PUF52
> Спасибо

## It is highly disrespectful and provoking that you continue to write in Russian, even after @paulfertser asked you to use English!

These kind people **deserve** to be respected, since they provide us with such advanced software for free, so we should at least do our best in terms of comprehensibility, don't you think?

A very nice way of achieving relatively good translation is using [this](https://ru.pons.com/%D0%BF%D0%B5%D1%80%D0%B5%D0%B2%D0%BE%D0%B4-%D1%82%D0%B5%D0%BA%D1%81%D1%82%D0%B0). 😊

thanks in advance
kind rgards

PUF52, 2022-01-13:
Sorry, I didn't mean to offend anyone, but Russian is my native language, English is not. It seems to me that anyone can use google to translate text in my language. As, however, I do when reading your texts. Sorry again.

discip, 2022-01-13:
Thank you!
My native language is also not English, but since the developer prefers to use English, we should respect this!

kudos for reacting objectively 👍

DimaStebaev, 2022-01-13:
Hey @Ralim.
I disassembled my iron and took a picture. Is the photo useful?
[image]

paulfertser, 2022-01-13:
On Thu, Jan 13, 2022 at 11:32:26AM -0800, Dima Stebaev wrote:
> I disassembled my iron and took a picture. Is the photo useful?

Yes, it shows that the PCB is marked v2.51A (so should be the same as
all the irons shipped for few years now) but the MCU is from
GigaDevices rather than ST (due to chip shortage). It's supposed to be
a direct replacement but since it's an independent implementation it
behaves a bit differently in certain edge cases.

I've seen people mentioning that GD32 didn't have the I2C bugs that
were present on all STM32F103 devices, and that the workarounds can
actually make the code unsuitable for GD32.

@Ralim is waiting for delivery of an iron with this MCU, and once it's
in his hands I'm sure it won't take long to make the firmware
compatible.

Thank you for reporting.

paulfertser, 2022-01-13:
On Thu, Jan 13, 2022 at 10:59:49PM +0300, Paul Fertser wrote:
> I've seen people mentioning that GD32 didn't have the I2C bugs that
> were present on all STM32F103 devices, and that the workarounds can
> actually make the code unsuitable for GD32.

@Ralim, found a possibly related note:

"In the ISR mode you shouldn't set BUFIE and/or POS bits before
sending the address, or else it leads to an endless loop in the ISR
handler as SB isn't getting cleared"

I can't tell if this applies to HAL code that's used to handle I2C or
not, a quick look made me think it's more likely not.

PUF52, 2022-01-14:
Hello everyone! Got a soldering iron. Firmware version 2.20, DFU 3.48. Everything is working. I risked flashing at 2.16 from Ralim. Does not work, the behavior is the same as that of Dima Stebeev. I rolled back to 2.20 again, it works. Waiting for news from Ralim. Thank you.

Ralim, 2022-01-14:
My unit has arrived, going to give it a proper look at over the weekend as much as I can.
Looks like the issues are mostly constrained to the different I2C engine.

I tried turning off DMA and it looked to fix up most of the issues for my unit here, if you have time (and patience) if you could try loading the firmware from this action: https://github.com/Ralim/IronOS/actions/runs/1686756257

And just give it a quick test to see how well it works / whats broken / does it work at all

@paulfertser I think you have found what I'm heading towards, any idea where you turned up that nugget of knowledge?
Look to be running into grief in ST's hal for the I2C DMA accelerated write calls

paulfertser, 2022-01-14:
On Thu, Jan 13, 2022 at 11:15:48PM -0800, Ben V. Brown wrote:
> ***@***.*** I think you have found what I'm heading towards, any idea where
> you turned up that nugget of knowledge?

On the third page of google results for "gd32f103 i2c stm32f103" there
was a link to a web forum where people discussed it in russian:
https://electronix.ru/forum/index.php?app=core&module=members&controller=profile&id=49725

> Look to be running into grief in ST's hal for the I2C DMA
> accelerated write calls

The forum post says only reads are problematic, and also apparently
the author was not using DMA.

Gianluxx, 2022-01-16:
> I tried turning off DMA and it looked to fix up most of the issues for my unit here, if you have time (and patience) if you could try loading the firmware from this action: https://github.com/Ralim/IronOS/actions/runs/1686756257

Works good on mine with GD32! Only problem, maybe already existing also for stm32, is the bugged screen with Auto rotation.
[image]
[image]

PUF52, 2022-01-17:
Hello Ralim. Let me remind you, I have a new soldering iron, firmware 2.20, DFU 3.48 and problems with loading firmware 2.16. I tried to download the updated firmware I2C_Wrapper.cpp, but something goes wrong. The firmware does not load, the disk disappears, the computer writes that the disk is no longer on the computer. I can't figure out what's wrong. A rollback to 2.20 restores everything and everything works correctly.

stranger271, 2022-01-24:
> Hello Ralim. Let me remind you,  

The Chinese have switched to a new microcontroller. I hope Radim can handle it. I can't do it without a turbo button

discip, 2022-01-24:
Good afternoon @stranger271,
> The Chinese

This might get understood the wrong way, you should say **`miniware`** instead.

>... have switched ...

Someone might also not get it the way you probably intended it to come along.

They did this not out of the blue, this decision had to be made due to chip shortage as pointed out here:

https://github.com/Ralim/IronOS/issues/1177#issuecomment-1012462721_

Ralim, 2022-01-24:
> Hey @Ralim. I disassembled my iron and took a picture. Is the photo useful? [image]

Thank you for confirming this, and taking a lovely crisp photo.

Confirms expectations, and I think we can say anyone on DFU 3.48 has been "upgraded" to the `GD32F103`.

It should be safe to run any new firmwares for testing and then roll back as far as I know.

My hold up is that my unit here seems to have lost its ability to reflash 🤣 (I use an external programmer, suspect something went wrong). So trying to ~unf$ck~ unbrick it at the moment.

In the mean time, if anyone who is keen can try the files from the github action i linked [here](https://github.com/Ralim/IronOS/actions/runs/1686756257), would love testing to figure out what else is broken other than screen rotation.
Just go to that page, and download the ts100.zip to get the files.

-----------------

In terms of languages, part of the problem is that I only speak English. But as you said, google translate does work well and thats what I use. The MAIN issue is that Github search does not translate. So anything said in any other language cant be searched. And given that already about 30% of the github tickets and noise is from duplicates, keeping to one language makes it easier for people to search so they open _less_ duplicates.

discip, 2022-01-24:
> So anything said in any other language cant be searched.

Very good point! 👍
I did not even think about this highly crucial one!

piksel, 2022-01-27:
Also, on the language note. You can translate the entire page to another language, but if every comment is in a different language you have to switch the source language between comments and it get's very hard to follow, not to mention quote!

KALLIAK, 2022-01-29:
> In the mean time, if anyone who is keen can try the files from the github action i linked [here](https://github.com/Ralim/IronOS/actions/runs/1686756257), would love testing to figure out what else [...]

Hello @Ralim! How can I download this zip? Github says than "This artifact has expired..."

Ralim, 2022-01-30:
Hello,
All build artefacts expire after a bit as free storage isnt infinite :)

But, you can download it from here https://github.com/Ralim/IronOS/actions/runs/1767487554

Have made #1202 of where I've gotten to.
Havent been able to make meaningful progress on getting DMA working, so fallen back to IRQ mode instead. Should perform about the same as far as users can tell.

If anyone can give it a test would be ideal.

KALLIAK, 2022-01-30:
Just tested on TS100 with DFU 3.17. Firmware works like intended as far as I can tell

PUF52, 2022-01-30:
On DFU 3.48, none of the firmware is installed. Only Chinese version 2.20.

KALLIAK, 2022-01-30:
> On DFU 3.48, none of the firmware is installed. Only Chinese version 2.20.

Are you sure that you flashing specifically [this](https://github.com/Ralim/IronOS/suites/5110180420/artifacts/153162185) firmware, and not 2.16 in master?

PUF52, 2022-01-30:
Sure

Ralim, 2022-02-03:
Did you flash the hex file after unzipping it?
Did you get the .RDY file ?

PUF52, 2022-02-03:
Шестнадцеричный, RDY не получил. На кмпьютере исчезает диск паяльника и он пишет, что этого диска больше нет на компьтере. Китайская версия 2.20 встает корректно, ничего не исчезает и процесс прошивки проходит штатно.

discip, 2022-02-03:
@PUF52 
Please use english.

PUF52, 2022-02-03:
I'm sorry I forgot

PUF52, 2022-02-03:
Hexadecimal, RDY not received. The soldering iron disk disappears on the computer and it writes that this disk is no longer on the computer. The Chinese version 2.20 is delivered correctly, nothing disappears and the firmware process runs normally.

Ralim, 2022-02-03:
Are you getting a .ERR file?
If you are you may just need to copy the file a second time.
The bootloader is somewhat well known for being buggy with some systems.

PUF52, 2022-02-03:
And this is how the soldering iron behaves on 2 different computers and with different USB cords.

PUF52, 2022-02-03:
I don't get the Err file. Just a soldering iron falls off the computer

Gianluxx, 2022-02-03:
Been using it for 18 days with the Actions firmware. All good except the already told rotation screen bug.

Ralim, 2022-02-04:
@Gianluxx  I have not been able to reproduce the screen rotation bug here.
Any chance you could test a reflash of the latest master for me ?

Gianluxx, 2022-02-04:
Same issue. Only in soldering mode. In settings it shows normal

Ralim, 2022-02-05:
Okay can you try [this action](https://github.com/Ralim/IronOS/actions/runs/1798021042)
I think I've spotted the bug :)

Gianluxx, 2022-02-05:
Perfect now!

tichy158, 2022-02-05:
> Okay can you try [this action](https://github.com/Ralim/IronOS/actions/runs/1798114842) I think I've spotted the bug :)

Hi, today the TS100 arrived and I tried to flash the firmware and unfortunately without success.
DFU: 3.48 (on display) (I assume I have a version with gd32f103)
I will connect the TS100 to a USB port in DFU mode. I will copy the file and the disk will be disconnected, but will not be reconnected. After manual disconnection and USB connection only shows a blank disk. The firmware does not work and I have to load MINI220.hex back and everything is in its original state. I also tried the official release and the release of action from this thread, nothing worked.

Gansgod, 2022-02-05:
i have the same problem! no other firmware is installed on dfu 3.48. only 2.20

Ralim, 2022-02-05:
@Gansgod Can you please eplain in more detail?
I'm testing here with 3.48 without issues.
Most likely you are running into issues that are _not_ to do with this code repository.
If the file is not loeading onto the disk and giving you a `.RDY` file, its not the firmware.

tichy158, 2022-02-05:
I tested on several computers with different cables and the behavior is the same everywhere. I will upload the firmware and the copy will not work properly, the disk will disconnect and no longer connect. When the power is turned on, nothing is displayed and the firmware does not work. I connect to the same port and with the same cable, I upload MINI220.hex and everything goes as it should. The file is loaded, the disk is unmounted, then mounted and showing the .rdy file. Maybe there is a problem with the flash memory mapping? Personally, it occurs to me that the bootloader stops writing and causes an error that leads to a processor reset (something like coore panic ...?) And does not even return to DFU mode. As a result, part of the firmware is already written but not complete, and this will cause a malfunction.

Ralim, 2022-02-06:
So firstly there is **NO** disk when running this firmware, this firmware does _not_ use USB to save code space.
So seeing no disk is normal when running the firmware.e

So to confirm, after you flash the hex file you are seeing TS100_EN.RDY ? (Please test with the EN build first, as its the most QA'ed).

Flash memory mapping in the 3.47, 3.48 and 3.49 is the same as all the previous versions from what I've been told.

Now, a bug in their bootloader is _super_ common. Which is where the steps like double-copying the file, using a different pc etc.

Can you confirm exactly which file you are attempting to flash?

tichy158, 2022-02-06:
The file I use is a link: https://github.com/Ralim/IronOS/actions/runs/1798114842. Uploading file TS100_CS.hex. If I upload the TS100_CS.hex file, the TS100_CS.rdy file will not appear. However, if I upload the MINI220.hex file (original from the manufacturer), everything works as it should (the MINI220.rdy file will appear). I tried to use the TS100_EN.hex file and the result is the same. The unit disconnects and no longer connects, so I don't see rdy and err either. I also noticed that the display shows that the solder is in DFU mode, but I believe that this is only the last state before the processor reset and that the new firmware no longer starts, the inscription will remain unchanged.

Ralim, 2022-02-06:
Okay so most likely this is a computer <=> bootloader issue.

Please make sure you have already extracted the TS100_EN.hex and that you are copying using the windows built in file copy (dont use any other software).

If you are not seeing the .RDY file show up, it sounds like a bootloader bug or computer issue. Miniware's bootloader has plenty of bugs, especially on newer operating systems as FAT12 is being deprecated.

tichy158, 2022-02-06:
I tried copying on several computers (2 laptops and a desktop). I tried various USB (USB 3.0, USB 2.0, on board USB, expanded USB ...) and two cables on the desktop. I always used an unzipped file and windows drag and drop function. The result was always the same. TS100_CS.hex never worked and MINI220.hex always worked. Maybe the problem is caused by the file size and the DFU ends up with some error. This is probably a bug in the bootloader. How much memory does the original processor from STM have? The gd32f103TBU6 replacement has 128K flash memory. The question is whether it's really this chip or is it some other model with lower memory and it's fake.

It's two o'clock in the morning, I'm going to bed :-).
Thanks for all the answers.

Gansgod, 2022-02-06:
@Ralim When I try to flash, I get this error.
And when flashing native firmware, everything goes smoothly
attached video
https://youtu.be/zfGZVyvOgyc

discip, 2022-02-06:
As proposed [here](#1214) already, maybe you could try the [DFU bootloader](https://github.com/Ralim/IronOS-dfu/suites/5187999152/artifacts/157853503) by @Ralim.

Before doing anything though read the docs carefully [Readme](https://github.com/Ralim/IronOS-dfu/blob/mainline/README.md), [Backup](https://github.com/Ralim/IronOS-dfu/blob/mainline/docs/BackUp.md) & [Bootloader](https://github.com/Ralim/IronOS-dfu/blob/mainline/docs/Bootloader.md)

He tried it on his **TS100**, **TS80** & **TS80P** and I also tried it on my **TS80P**.
I am currently running the latest build successfully!

PUF52, 2022-02-06:
As I wrote earlier, I don’t dare to flash the bootloader from Ralim myself, I’m afraid to screw up the soldering iron.
--
Отправлено из Mail.ru для Android воскресенье,

jhamhader, 2022-02-11:
> Okay can you try [this action](https://github.com/Ralim/IronOS/actions/runs/1798114842) I think I've spotted the bug :)

Tried the firmware from this action with TS100 DFU 3.47 and it worked. Waiting for the next release :)

Gansgod, 2022-02-13:
I managed to solve the firmware problem on my TS100 soldering iron with DFU 3.48 (many thanks to the user "JugglerLKR", he suggested what to do)
You need to install UBUNTU 18.04. It is on this version that the soldering iron is flashed without problems.
If someone needs explanations in Russian, you can write to me in telegram - Egor_138

wiebenor, 2022-02-14:
I am glad I'm not running only DFU 3.45 then, as well as presumably the older chip, because this problem, or problems, even just as a bystander, gives me a headache to read... Much love for Ralim and those others who help make this IronOS work so flawlessly, and troubleshoot problems like these

Ralim, 2022-02-14:
I _think_ we are ok to close this out now? 😓

alwwwex, 2022-02-14:
Hello! I have totally same behavior as @Gansgod here:
https://github.com/Ralim/IronOS/issues/1177#issuecomment-1030749725

DFU 3.48 and GD chip, have tried to use Win 10 Pro, Win 7 - same issue with last 2.17 - still going to brick and only original 2.20 can be easily uploaded :(

I have also tried to use CMD Copy function instead of win explorer UI. Will try on some linux machine...

Ralim, 2022-02-14:
Hello,
I'm suspect there is a bug in the Miniware bootloader for these devices somewhere. I've asked them about it but no response yet. I'm guessing either windows has updated or they have made an assumption that doesnt hold on larger firmware.

To confirm, what DFU version are you using?

alwwwex, 2022-02-14:
I have bought TS100 in 2022 and there was DFU 3.48 and GD microcontroller. Now, I have successfully upload last 2.17 hex using Raspberry (some 2021-01-11-raspios-buster OS). First time it have showed file with .ERR (never even seen that on Win) and then, have upload file one more time (.ERR was still in folder) - after reboot, there was only RDY file and now its works. Interesting, that original file somehow was working as should be on whatever system, only custom one causes an error of DFU + second question, what is difference between Win and Linux at this point...

Yachtie2020, 2022-03-25:
Hey, I fought with this for awhile, but got it working finally. I am posting details of exactly what I did to get it to work so others can follow, see Work Around / Solution. This solution has been mentioned before, but I did it in a way it initially failed on the suggestions I found.

Problem:
My TS100 is DFU 3.48 which means it is likely a GD micro (I did confirm this). Like others, on Windows flashing it fails and disconnects after copying the file to the root level of the TS100. I can reload the original 2.20 version from manufacture to get the TS100 back and running. I can load version 2.14, 2.15, but the screen / OLED is messed up, but the TS100 does start to get hot, so it is partially working.

I tried following issue #1214 and mis read some bits that made me think bad OLED connections might be related to flashing, but after expanding all posts and re-reading it was clear there where 2 separate issues for that ticket. #1214 does mention Ubuntu a few times.

Work Around / Solution for me:
You need Ubuntu 18.04.6 LTS Bionic Beaver as mentioned above. Follow guides on internet to make a bootable CD/USB. **Using the "Files" program(s), drag the hex file to the TS100 root folder (Hold button near hot end while plugging power to boot into DFU (Flashing) mode). First time this failed with .err file, 2nd time it worked with .RDY! I did not delete the .ERR file, I just dragged file between Files windows again and had 2 files a .ERR and .RDY**

Other information:
I tried the Raspberry Pi option with my Zero W model (headless), but the linux script failed to copy with similar issue to Windows, I also had issues with the script hanging sometimes waiting for TS100 to remount. The TS100 had to be unplugged and replug to re mount and re-attempt. I also had to umount the TS100 initially as it was getting auto mounted and causing script to error on already mounted error. I tried the same thing on Ubuntu live using the script and had same issue. I was also having heaps of issues with the Desktop auto mounting the TS100 on plugin and during the script execution, specifically after loading firmware and sleep 5 seconds I would see Ubuntu auto mount it again. I am assuming the script is unmounting and the desktop is auto mounting it straight away causing the next mount in script to error.

I tried modifying script to unmount the auto mount before each mount in script as work around and was about to figure out how to disable unmount when I realised I saw a comment about just drag the file on Ubuntu like on Windows and it clicked to try without script.

It is possible the Raspberry Pi option will work if not using script, I haven't tested, I blew enough time trying to get a Pi working / connected to Wifi to discover the latest Raspberry OS has a known DHCP issue out of the box...

Hopefully the Linux script information might help provide information, as I'm not sure if anyone else has discovered the issue exists on Ubuntu using the script. Is there a race condition and/or mounting issue going on?

Marinaxs, 2022-04-13:
I got my TS100 today with 2.20 and DFU 3.48
Same Issue on windows with deconnecting
so i try Ubuntu - same problem but now i stay connected an try several times -> .err
Now it worked with ralim

[18 more replies]

closed 2022-04-21 by Ralim