Ralim/IronOS #1182: need real documentation, description of menus, effects of selections, troubleshooting power issues, etc.

cm68, 2022-01-17:
the documentation is very fragmented, and there is no single document that describes this wonderful software's features.

the right way to do this is to have embedded in the source code structured comments that are used to generate markup,
like doxyen or the like.   this will make it easier to keep the doc and the code sync'd up.

either way, there ought to be a manual, describing the menus and what all the selections do.

Ralim, 2022-01-17:
The intent is that the documentation is _in_ the OS. This is why there is the pause-to-see-help feature on all of the menu options.

Have meant to write an "overview" page that goes through most of the core features of the menus, but generally the TL;DR is just you press front button to get hot, and rear button to enter settings menu. Then use the settings help to check what settings do.

Internal code comments should be enough for most things I would hope. Definitely do not like Doxygen as it tends to just be a "here is a fancy way to read the same comments your IDE is going to show you anyway" and its a bunch of overhead to support and maintain.

cm68, 2022-01-17:
i understand the sentiment.  doxygen is not the ideal system, it's just better than cweb, which is UGLY.

the documentation that gets displayed on the interface tells what, but not why.

for example,  why would I want to adjust the PD timeout? no idea.  I have two power banks that claim to do 15 and 20 volt PD, but I don't seem to be able to get them to use more than 12V.  how to debug this?  it's this kind of question that needs to be in a doc.

keeping a codebase and documentation in sync is an unsolved problem, and javadoc kinda handles this for java, but there is no consensus on c/c++.

Ralim, 2022-01-17:
>for example, why would I want to adjust the PD timeout?
Fair point. Its only intended to be changed if you _know_ you need to change it :)
-> To answer the question, its for when QC chargers are being a pain and requiring a fast negotiation.

> I have two power banks that claim to do 15 and 20 volt PD, but I don't seem to be able to get them to use more than 12V

As PD is aware of the chargers rating, for PD it will only negotiate as high a voltage as it can _safely_ without violating the chargers max current rating. So it picks the highest of (Device max voltage, charger max voltage-considering tip resistance and checking it to the current rating)

Without knowing the devices involved I cant comment more than that :)

I think what this repo really needs is just an in depth menu explainer. Not really a code docs system.
I'll have a think about how to automate this

cm68, 2022-01-18:
your software is great; your code hygiene is your business.  respect.  both of the power banks I have claim to do lots of current, so I'm not sure about the negotiation you describe.

[image]
[image]

Ralim, 2022-01-18:
I will take it onboard to add more documentation to the main repo though.

The TS80P only supports 12V as the maximum voltage so you are correctly seeing it negotiate the max voltage supported.

closed 2022-01-24 by Ralim