Ralim/IronOS #1191: Setting USB-C Negotiation Voltage in v2.16 on Pinecil

sklepa, 2022-01-21:
Is there a setting that I can choose negotiated voltage over powering via USB-C? There was such feature in my first OS version which was 2.14. My power bank is capable of 12V and it worked on v2.14. Now i updated firmware to 2.16 polish lng. and iron negotiates only 10V(strange value, ADC is calibrated). Am I forseeing something?
Cheers

Ralim, 2022-01-21:
No IronOS version has supported you changing the PD voltage. Only the QC voltage can be changed.

Make sure you are using a C-C cable connected to a USB-PD capable usb-C port.

You can also verify if it's using QC or PD in the debug menu (hold - on home screen until it shows up, press + to go through readouts)

Ralim, 2022-01-21:
For PD it uses the tip resistance (around 8 ohms) and the power supplies advertised voltage and current settings to select the highest voltage possible while keeping peak current below the rating of the power adaptor.

A very low wattage PD adaptor could negotiate 10V

sklepa, 2022-01-21:
Yes I mean QC voltage, there was such option in 2.14 and i was able to set 12V and it worked on USB-A to C previously.
In debug menu(when connected A to C) PWR shows PD then after couple seconds QC for a second and back to PD :o strange. On C-C the "PWR PD" is stable. 
Power bank supports QC, this is 18W baseus 10Ah

Ralim, 2022-01-21:
Most likely your powerbank only supports QC if negotiated quickly.
Try turning _down_ (or to 0) the PD timeout variable to force the PD negotiation to stop earlier on.

Note for Pinecil QC is not reccomended as it generally only does 12V.

sklepa, 2022-01-21:
Turning down time does not helps. But anyway, Thank You very much for clarifying topic :)
I will use it via C-C connection. Of course on 12V it's not performing it's tops, but this is what I have, and for mobile use it makes a very portable set. For grid connection I managed to buy very sexy delta PSU ;>
Thanks for Your work! Greetings from snowy poland!

closed 2022-02-04 by Ralim