Ralim/IronOS #1192: TS100 not heating up after powering on

Saschko2000, 2022-01-21:
**Describe the bug**
I set the iron to startup in soldering mode.
After powering it shows the actual temperature but does not heat up. After some seconds it shown "thermal runaway".
After some more seconds it starts to heat up an works flawlessly.

**To Reproduce**
I just need to switch off and switch on again.

**Expected behavior**
After powering the iron should head up to the desired temperature

**Details of your devices:**
I use a 24V Powersupply
DFU 3.4.3
Release 2.16
Everything is working correct with Release 2.09
Tested with two different irons. Both behave the same.

Ralim, 2022-01-21:
So thermal runaway is trigger if after ~3 seconds it doesn't see the tip temperature increase.

What power source are you using (can it supply the current)?

I test here on a ts100 at 20V with no issues.

What top are you using? 
What setpoint temperature ?

Have you tried a factory reset?

You should hit 200C within the 3s thermal lockout so I have no idea how you can trigger the warning

Saschko2000, 2022-01-21:
I tested again with a 4s Lipo. And a third iron (my father'S)
The behavior is the same with all three of them.

**What I did:**
- flashed firmware 2.16
- Factory Reste
- Set startup behavior to soldering mode.
- power off
- power on again.

**Behavior:**
- They show the Temperature but did not heat up.
- after 23 seconds they show Thermal Runaway
- after another 10 seconds they show the icons for heat up and menu
- after pressing the heat up button everything works as expected

All three behave the same. Two are from 2018, and one from 12/2021.
All three work perfect with version 2.09

Can I do anything else to help finding out the Problem?

Firebie, 2022-01-21:
Is it possible that battery voltage drops too much?

Saschko2000, 2022-01-21:
Then the voltage drop must accour on two different powersources, and the battery. And the battery is one of my copter 4s 2000mah.
So I think it is very unlikely. 
And it would not explain, why it works on version 2.09.

Saschko2000, 2022-01-22:
I have tested to heat up the tip externaly. 
Then it looks like the iron reads the temperature 3 to 4 times after power on, and then stops.
Then the same:
- It shows the Temperature but does not heat up.
- after 23 seconds it shows Thermal Runaway
- after another 10 seconds it shows the icons for heat up and menu
- after pressing the heat up button everything works as expected

Hope that helps

Ralim, 2022-01-24:
>Set startup behavior to soldering mode.

This is the most likely cause, as this is not a well tested or used mode.
I'll have a look into it, most likely this is an older bug that is only now being noticed because of the thermal runaway protection.

discip, 2022-01-25:
I second this finding.
For me a similarly faulty behavior occurs with my **TS80P**.
However the **Pinecil** works as expected

device|soldering **`S`**|sleeping **`Z`**|hibernating **`R`**
-|-|-|-
**TS80P**|&#10060; *after **~23s*** *Thermal&nbsp;runaway*|&#10060; **\|** &#10004; *after **~23s*** * |&#10004;
**Pinecil**|&#10004;|&#10004;|&#10004;

\* The iron does not heat up to the sleep temperature immediately after being powered on.
Only about 23 seconds later it does this.
If you trigger the accelerometer before that time nothing will happen.
However if that time has passed by, the tip reached the sleep temperature and you move the iron, it than heats up to the soldering temperature.

hazymat, 2022-02-03:
I came searching for this as I have the exact same issue.

More info:
- I have tested with 2 different TS100 irons (from different sources, reputable UK seller and Aliexpress)
- Both have worked for a couple of years fine on a previous version of IronOS
- My power supply is a Meanwell 24V / 5A supply which I spec'd especially for TS100, using nice silicone wires and a barrel jack all rated above 3A.

When I first upgraded both irons, I used them a bit, played with the new menus, etc. Both irons heated up fine and I set them up as I liked them. Then I left for about an hour, came back, plugged one in, and it wouldn't heat up. But shows the current temp (room temp) fine. Switched to the other iron, exactly the same problem.

Only on one occasion did I see the "thermal runaway" message. On another occasion it heated up but extremely slowly (1 degree C every second). After multiple tries neither now heat up.

Factory reset did not resolve.

Downgrade to previous version of Ralim IronOS and it works fine again.

Ralim, 2022-02-04:
Hia,
Could you test the files from this build : https://github.com/Ralim/IronOS/actions/runs/1797609654

Think I've found the bug in the fast/slow pwm changeover

discip, 2022-02-05:
@Ralim
## Now it works as intended! 😃👍
Thank you

Ralim, 2022-02-05:
Going to close this as working; please yell if its not fixed for you :)

closed 2022-02-05 by Ralim

Saschko2000, 2022-02-05:
Thanks very much for bugfixing this. 

Where can I find 2.17 ? 
May be I am to blind to see :-(

discip, 2022-02-05:
@Saschko2000
This is planned to be released some day.
The latest build is [here](https://github.com/Ralim/IronOS/actions/runs/1798190200) *(Look for the build appropriate for your device on the bottom of the page.)*!

Saschko2000, 2022-02-05:
Thank you vera much.

Some day in the future, i will have to learn to scroll 😊