Ralim/IronOS #1193: Add Hall Effect Sensor to Debug Menu

ritkit, 2022-01-21:
**Is your feature request related to a problem? Please describe.**
I am looking to debug the hall effect sensor that I added to my Pinecil. I want to see how sensitive it is, how impacted by the soldering tips, and design a holder that can reliably trigger it. 

**Describe the solution you'd like**
I would make the following changes in the code and it should be pretty straightforward.

**GUIThread.cpp**
```
void showDebugMenu(void) {
	...
	switch (screen) {
		...
		case 13:
			// Print raw hall effect value if availabe, none if hall effect disabled.

			if (getHallSensorFitted()) {
				getRawHallEffect()
			}
			else {
				OLED::print("None", FontStyle::SMALL);
			}
		...
	}

    ...

    if (b == BUTTON_B_SHORT)
      return;
    else if (b == BUTTON_F_SHORT) {
		screen++;
		screen = screen % 14;
    }
	...
}

```

**make_translation.py**
```
...
def get_debug_menu() -> List[str]:
    return [
        datetime.today().strftime("%d-%m-%y"),
        "HW G ",
        "HW M ",
        "HW P ",
        "Time ",
        "Move ",
        "RTip ",
        "CTip ",
        "CHan ",
        "Vin  ",
        "ACC  ",
        "PWR  ",
        "Max  ",
        "Hall ",
    ]

```
**Describe alternatives you've considered**
I cannot think of any off the top of my head, but that is why I am putting this FR here to ask for input.

**Additional context**
I have not really made any major contributions to public repositories and want to really start cranking that up. Thought this would be a great first start, newbie, project. Wanted to put in my two cents and see if the team agrees with the input. Please feel free to correct this FR or any procedural items!

Ralim, 2022-01-21:
Hi,
Please submit your changes as a pull request if possible.

All of that code looks good except for the `OLED::print("None", FontStyle::SMALL);` as you need to use a translated string in the `OLED::print()`. But I'm happy to help with that.

ritkit, 2022-01-21:
Oh ya good point, will dig through how you implement translations, I did see its a combo of JSON. So I will take care of that. I dropped the code here, just to make sure I am sane. Will do the same with the translation changes if you do not mind.

ritkit, 2022-01-21:
Never mind, that was awesomely easy! I will just swap "None" for translatedString(Tr->OffString). No reason to add another new message to the translations.

Ralim, 2022-01-21:
Yep that should be a workable string to re-use :)
And that way it automagically will work across languages as well which is nice.

Ralim, 2022-01-21:
Would prefer using a PR for this, if you can :)
If you cant let me know and I can create one later on tonight

federicodilo, 2022-01-23:
What about using "SettingNAChar"?

Ralim, 2022-01-23:
N/A has been removed in newer builds as it was not being used hmmm

federicodilo, 2022-01-23:
I'm afraid that "OffString" may mislead the user who thinks of an Hall sensor already inside the iron but just turned off.

closed 2022-01-25 by ritkit