Ralim/IronOS #1198: Cant flash the firmware

tdboots, 2022-01-25:
hi i try and flash the firmware and when i do all my usb devices disconnect and it says there is an unexpected error, i don't know what it could be.

Ralim, 2022-01-25:
What device?
What operating system?
Have you searched all the existing issues around flashing?
Have you tried another computer?

tdboots, 2022-01-25:
> 

hi its the ts100 windows 10 I've tried 2.16 and 2.15 and I couldn't find any issues that's why I created an issue and this is the only computer I've tried on.

mosch, 2022-01-26:
same here, no luck. only getting the .err file. tried on win 10 and osx.

DFU: 3.45
Original FW Ver. 2.18

Ralim, 2022-01-27:
Make sure you are trying with the `.hex` file.
Otherwise its mostly down to your specific computer and software you have installed.

other issues include #11 #298 and _many_ more

closed 2022-02-04 by Ralim