Ralim/IronOS #1199: Firmware does not work on dfu 3.47 on ts100

Cupitori, 2022-01-26:
**Describe the bug**

After flashing v2.16 on TS100 with DFU 3.47 the iron becomes a brick.
**To Reproduce**

**Expected behavior**

**Details of your device:**
 
 Device: TS100Release: 
Can't get hardware release, late 2021 model, DFU 3.47
Power adapter being used: YH24-3

**Additional context**

After flashing the iron still enters DFU mode.
Default firmware v2.18 does not work as well and causes the same behavior.
immediately 2.20 was not flashed, I flashed the firmware in order from the oldest to 2.20
Default firmware v2.20 works correctly. Device is work.

discip, 2022-01-26:
Good evening @Cupitori.
# Please search in the already existing issues before opening a new one. 😊

If I am right, then you are at least the 4th one addressing this topic!

https://github.com/Ralim/IronOS/discussions/1155#discussioncomment-2032230
https://github.com/Ralim/IronOS/issues/1170
https://github.com/Ralim/IronOS/issues/1177

Ralim, 2022-01-26:
Closing this issue as its a duplicate of existing ones and notes on the release page.

closed 2022-01-26 by Ralim