Ralim/IronOS #1200: Flashing screen for confirmation or notification

River-b, 2022-01-26:
Problem is you don't know when the pinecil goes to sleep especially if you have it mounted in a holder for you and are using your two hands to hold the wires.

Solutions:
(1) piezo beeper mod  by end user  that has audible beep when you make changes or there is state change like pinecil goes to sleep on you.
(2) a way to add an small LED light that flashes a number of times to indicate things
(3)  Simply flashing the screen a number of times to indicate things, examples:
       - pinecil goes to sleep, flash 3-5  times all white screen so it's really clear
       - you just pressed Boost successfully, 2 Flashes of whole screen with words SUPERBOOST
       - you successfully updated Firmware,  FLASH a couple times version # like,    SUCCESS! v. 2.16

Additional notes, the small white LED, could also be used as a flashlight at front of the case, but then have additional feedback codes to do things above.

prokrypt, 2022-01-27:
Yes! Please add beeping/flashing morse code status output!

River-b, 2022-01-27:
interesting idea flashing the screen all white light in morse code. like long white screen, and short flashes.  I will watch tutorial to learn morse with flashlight. 
what message would you want in Morse as opposed to it just flashing all white inverse screen a  few times to let you know Pinecil is  sleeping or Pincil state change is awake now?

I'm not sure they can add beeping right now as that would mean a hardware/pcb change. but the  Flashing on state changes and other stuff might be doable?

discip, 2022-02-03:
@prokrypt 
> ... beeping/ ... morse code status output!

@River-b
> interesting idea flashing the screen all white light in morse code. like . . .

## Are you being serious about that guys? &ensp;Morse code? &ensp;For real? 😲
How long do think will it take you to decipher the code for: **`"350°C Ready to solder!"`** compared to a quick glimpse on the screen displaying the same thing? 🤦‍♂️

This feature might be worth the effort for a device featuring only one LED as output!

River-b, 2022-02-09:
I am not serious about morse code , but I think Prokrypt is being serious as he is a ham operator lol

discip, 2022-04-21:
@River-b
Please close this issue, since this repo is mainly focused on the software side of already available devices.
So no hardware mods supported. You can not attend everyone's wishes. That would get out of hand very quickly! 😊

thanks

River-b, 2022-04-22:
none

closed 2022-04-22 by River-b