Ralim/IronOS #1201: Feature request on MHP30

ltwin8, 2022-01-27:
Feature request is not related to a problem.

Would it be possible (since a Hortplatz doesn’t need all the settings a solderingiron does) to have a menu with for example 5 points consisting of temperature and then times between, example point 1 (say 111degC) to point 2 (say 200 degC) and a time-Frame of 50s

Time = 0s (first points temperature reached, plate will try to reach this temp as fast as possible)

Calculate time (linear) (point_n - point_n-1) / timeframe so (200-111)degC/50s is 1.78degC per second

MCU Do this for the next pints the same, should be okay to have one float per timeframe to store slope

Now maybe you start a timer at the moment Point1 temperature is reached, now only tick every second for example and update new Setpoint temperature according to elapsed time x slope (of course only to the next point and then continue from there…

Sadly I do not have a MHP30 yet.

Hope it is understandable enough.

Keep on with that great work

rofi, 2022-09-27:
I would really love to have this feature!

nizce, 2023-03-05:
This would be great if it could be added!

codingcatgirl, 2023-04-22:
This would be a killer feature! Please don't tempt me to implement it myself i can already feel myself dragged towards the source code <.<

codingcatgirl, 2023-04-23:
Oops, too late, i've implemented it. Head to #1672 if you wanna give it a try.

ltwin8, 2023-04-24:
Wow you really don’t waste any time, great job, thank you for the work

nizce, 2023-04-24:
> Oops, too late, i've implemented it. Head to #1672 if you wanna give it a try.

Wow that was quick! Cant wait to try out the new feature when availible 😎

closed 2023-04-25 by discip