Ralim/IronOS #1203: Hall Sensor Sensitivity needs Tuning Calibration

River-b, 2022-02-03:
Hall Sensor even on H - High setting requires magnet that is very close & strong.

 - Device: Pinecil
 - Release: 2.16
 - Power adapter being used:  Pinepower or  variable 24V AC adaptor.

Need to tune OS settings for hall sensor so that High is more reactive/sensitive
I installed sensors on 2x pinecils and have a few different magnets on order.
I will install the PR OS tonight to test.

River-b, 2022-02-03:
Current settings:
- Installed PR  2.16.FBDD38A
- Test #1  
-    sitting in the metal stand with no magnet  Hall reads 1-2
   when the magnet is kissing the Pinecil right over the sensor, then it is around 2000 but it jumps around a lot even when it is touching pinecil on the sweet spot 1300-3000.  
then when I slowly pull magnet away from the pinecil,  , I can see it  drop to  0, but number after 0 is 65535

I will experiment more later.

Ralim, 2022-02-04:
Okay so now the menu is merged into master, so if you flash [this action or newer](https://github.com/Ralim/IronOS/actions/runs/1797355529) then it should be in the menu and should show the abs() of the field strength.

Currently thresholds are 1000,500,100.
We could probably change these to 200,100,50 perhaps?

River-b, 2022-02-04:
yes I think 50  for HIGH would be nice.  is there a test version of firmware with  50  for High  so I can install and test this? 
Maybe keep  1000 for low for people who have very powerful magnets or have a reason they want the hall to be weak.
400 medium
50 for high so it's super sensitive or most sensitive so you can use tiny earth magnet if you want or have hall sensor at different angles when you insert into Stand without needing to glue magnets in several clock spots around the stand?

River-b, 2022-02-04:
Ralim,  is there a version with  HIGH = 50 now that I could install to test out?   I have time to test it.

River-b, 2022-02-04:
is there a way I could modify firmware myself to be:
L= 1000
M= 400
H= 50  
then I could test it myself and modiy as needed.  but if you have a version with H=50 I could download that one?

River-b, 2022-02-05:
my goal for testing is to prove how low can I make HIGH sensing number before it causes issue with the Tip's own magnetic field triggering it.  I want to make sure that 50 will be safe to not trigger this.