Ralim/IronOS #1212: Add support for 28v PD EPR and vbus mod detection to the Pinecil

willlovesbearz, 2022-02-05:
EPR support with mod detection would be very nice for getting some extra power out of the pinecil, and should e possible to negotiate using the existing fusb302. it would not, however, be safe to utilize 28v on a normal unmodded pinecil due to the vbus pin, so epr should be locked behind a mod detection, which should be fairly simple to implement using the attach detection function of the fusb; it should register as detached if the pin is floating.

Ralim, 2022-02-05:
I'm planning to add mod detection.

EPR is a bit more tricky as there are not many EPR adapter's out there at the moment to test against :)

River-b, 2022-08-02:
Please close this ticket  as Pinecil V2 as of Aug 2.2022  now tentatively supports PD3.1 EPR  28V 140W.

thanos-the-tank-engine, 2022-08-03:
hm, fair enough. I'll be closing this issue as even though we're not at the point where you can just use an epr power supply with the pinecil we do at least have mod detection in there.

thanos-the-tank-engine, 2022-08-03:
.............................................................................................

closed 2022-08-03 by thanos-the-tank-engine