Ralim/IronOS #1213: Return to the problem of installing firmware 2.16 on the soldering iron TS-100 with DFU 3.48

PUF52, 2022-02-05:
Ralim, although you closed the problem with firmware 2.16 on DFU 3.48, I still have it and I don’t know how to solve it. No firmware is installed on the soldering iron, except for Chinese stock 2.20. When you try to install any other firmware, the soldering iron falls off the computer and the disk disappears.
On the other two soldering irons (backup) with DFU 3.45, any version of the firmware is installed, but with DFU3.48, nothing happens. I'm at a dead end. Please help.

PUF52, 2022-02-05:
Although everything on the computer is in Russian, but I think you will understand everything. Thank you.

closed 2022-02-05 by PUF52

discip, 2022-02-05:
@PUF52
If you are sure this is a bootloader issue, maybe you could try the DFU [bootloader](https://github.com/Ralim/IronOS-dfu/suites/5187999152/artifacts/157853503) by @Ralim.

Before doing anything though read the docs carefully [Readme](https://github.com/Ralim/IronOS-dfu/blob/mainline/README.md), [Backup](https://github.com/Ralim/IronOS-dfu/blob/mainline/docs/BackUp.md) & [Bootloader](https://github.com/Ralim/IronOS-dfu/blob/mainline/docs/Bootloader.md)

He tried it on his **TS100**, **TS80** & **TS80P** and I also tried it on my **TS80P**.
I am currently running the latest build successfully!