Ralim/IronOS #1214: Return to the problem of installing firmware 2.16 on TS100 with DFU 3.48

PUF52, 2022-02-05:
Ralim, although you closed this problem, it remained for me. Not a single firmware is loaded on the soldering iron, except for the Chinese stock 2.20, including your 2.16. after downloading and copying. I can not solve this problem myself and I ask for your help. On two other soldering irons with DFU 3.45, everything happens normally and without problems. I also loaded your corrected firmware - the behavior is similar, the soldering iron falls off the computer.

https://user-images.githubusercontent.com/86562320/152643926-54741d50-3938-4459-bd7e-053a178e74a2.mp4

discip, 2022-02-05:
@PUF52
Maybe try
1. removing any Cyrillic letters from the files name
2. copying the file using drag & drop

PUF52, 2022-02-05:
Replacing Cyrillic letters with English did not give anything. I'm afraid to flash a new alternative bootloader myself, I'm not such a great specialist, I'm afraid of a brick.

discip, 2022-02-05:
Seems you are not the only one facing this issue, so maybe some fellow sufferer is keen enough to try the custom bootloader for you. 😊

PUF52, 2022-02-05:
Replacing Cyrillic letters with English did not give anything. I'm afraid to flash an alternative bootloader myself, because I don't consider myself a specialist, I'm afraid of a brick.

PUF52, 2022-02-05:
Here is an interesting behavior of the firmware is visible. Now I remember the following: after receiving a soldering iron with firmware 2.20, I immediately flashed it to the Ralim 2.16 version and it behaved like Dima Stebev, a lot of artifacts, bugs, etc. Then, after modifications by Ralim (after requests from users) of firmware 2.16, I also flashed the modified firmware. And so, after that, leapfrog began - the soldering iron falls off the computer and no firmware gets on the soldering iron. What is it. The soldering iron itself somehow reconfigured. For the first time, 2.16, although it crookedly stood on the soldering iron, but then nothing. Unclear.

PUF52, 2022-02-05:
Ralim? good evening, and if I give you access to my computer through Team Vie
Ver, can you remotely change the bootloader for me?

Ralim, 2022-02-05:
Hia,
That error your are seeing is _NOT_ because of the firmware itself, but it is usually a sign of the bootloader having issues.

So far historically it has been fixed by:

* Different USB cable
* Different USB port
* Different Operating system
* Different computer
* Delete any failed .ERR or .OK files on the drive (but leave other files alone)

Most likely you have _something_ on your computer such as anti-virus that is interfering with the file copy.

Always make sure file names are in english and are kept short, as their firmware does not handle Russian characters and can cause issues.

You wont be able to change the bootloader either until you resolve this.

Sorry for the delay, timezone overlap means I'm not really around when you are much :)

PUF52, 2022-02-06:
Greetings. Tried all your suggestions, no progress. I understand that the firmware has nothing to do with it, it's all about the bootloader. How to downgrade DFU. I'm at a dead end.

PUF52, 2022-02-06:
By the way, I tried to install your modified firmware 2.16 on my backup soldering iron with DFU 3.45. Everything went well, the modified firmware got up successfully, which once again confirms that the problem lies in the bootloader. Is there a way to downgrade the DFU version?

JugglerLKR, 2022-02-06:
@PUF52 Do DFU 3.48 works as expected when flashing stock firmwares?

PUF52, 2022-02-06:
Of the stock, it only works correctly with v2.20, with which the soldering iron came.

paulfertser, 2022-02-06:
Hi,

Try flashing
https://github.com/Ralim/ts100/blob/master/Development%20Resources/force_blank_flash.hex
(it is going to do full erase of everything but the bootloader).

If this doesn't work (leads too bootloader rebooting) I'd try taking
just half of it and flashing again.

PUF52, 2022-02-06:
halfferzer, If I knew how it's done

JugglerLKR, 2022-02-06:
> halfferzer, If I knew how it's done

just flash it as you do with regular firmware. if it flashed ok (force_blank_flash.RDY), then flash Ralim's 2.16.

PUF52, 2022-02-06:
Send me this file, I can not understand and find it

JugglerLKR, 2022-02-06:
> Send me this file, I can not understand and find it

There is a link above https://github.com/Ralim/ts100/blob/master/Development%20Resources/force_blank_flash.hex

Download and flash it.

There is no PM on github so I reply in russian here:
Скачай файл по ссылке. Прошей в паяльник. Это файл для очистки флеша, но он не стриает dfu. Потом прошей прошивку Ralimа.

PUF52, 2022-02-06:
I can't figure out where to download this file.

paulfertser, 2022-02-06:
On Sun, Feb 06, 2022 at 06:15:27AM -0800, PUF52 wrote:
> I can't figure out where to download this file.

https://github.com/Ralim/IronOS/raw/master/Development%20Resources/force_blank_flash.hex

PUF52, 2022-02-06:
I can't find it, just send me the file

PUF52, 2022-02-06:
I couldn't find the download from your links.

discip, 2022-02-06:
@PUF52
Here you go.
[force_blank_flash.zip](https://github.com/Ralim/IronOS/files/8010426/force_blank_flash.zip)
️

PUF52, 2022-02-06:
ZIP file received, unpacked, copied to the soldering iron disk, but it is not put on the soldering iron, everything falls off, as when trying to install the firmware.

***@***.***

paulfertser, 2022-02-06:
On Sun, Feb 06, 2022 at 07:06:32AM -0800, PUF52 wrote:
> ZIP file received, unpacked, copied to the soldering iron disk, but it is not
> put on the soldering iron, everything falls off, as when trying to install the
> firmware.

Probably the filename does make a difference? Can you rename it to the
same name as "mini" firmware that works? Please try exactly that, and
with all the letters having the same case.

If the target still disconnects from the computer try cutting the file
in half (it's a text file, you can edit it and remove half the lines).

PUF52, 2022-02-06:
> Может быть, имя файла имеет значение? Можете ли вы переименовать его в то же имя, что и "мини" прошивка, которая работает? Пожалуйста, попробуйте именно так и со всеми буквами в одном регистре. Если цель по-прежнему отключается от компьютера, попробуйте разрезать файл пополам (это текстовый файл, вы можете отредактировать его и удалить половину строк).

Сделал, но не помогло

JugglerLKR, 2022-02-06:
> Сделал, но не помогло

Try booting from USB/CD latest Ubuntu and try flashing iron in it. If there is a problem with your windows installation, it should solve it. Also, as already mentione eralier, try different micro-usb cable (better use ones that comes with phones). Remove all other usb devices. Do not use hub/extenders. Do not flash from front USB, connect cable to rear usb (if your PC has both usb 2.0 and 3.0 use 2.0 first). Then, try another computer.

PUF52, 2022-02-06:
Tried everything - to no avail

discip, 2022-02-06:
Still assuming the bootloader is the culprit. 😉

PUF52, 2022-02-06:
I agree

JugglerLKR, 2022-02-06:
In the video iron disconnects while copying file.. This means flashing stock 2.20 from miniware also will not work.

PUF52, 2022-02-06:
Stock sews beautifully

JugglerLKR, 2022-02-06:
> Stock sews beautifully

Try to rename Ralim's firmware to exaclty the same as stock filename.

PUF52, 2022-02-07:
Failed. It doesn't solve the problem

Ralim, 2022-02-07:
[TS100_EN.zip](https://github.com/Ralim/IronOS/files/8012541/TS100_EN.zip)
Can you try this adjusted hex file?

Gansgod, 2022-02-07:
@Ralim Tried your file just now. The result is the same. Error. Soldering iron is not stitched

PUF52, 2022-02-07:
Tried. Didn't solve the problem. The soldering iron falls off.

Ralim, 2022-02-07:
As in it shows you the error or you get a .ERR file?

PUF52, 2022-02-07:
It's just that the disk that appeared after connecting the soldering iron to the computer disappears, the .err file does not appear.

Gansgod, 2022-02-07:
when you try to throw off the firmware file, the drive is turned off. and the soldering iron turns into a brick. Chinese firmware 2.20 is installed normally

PUF52, 2022-02-07:
This is what is not clear. Why does firmware 2.20 like DFU3.48 and it is successfully installed on a soldering iron, but any other firmware does not like it. Maybe it's in the firmware and DFU? in their incompatibility.

Ralim, 2022-02-07:
I have tested here with DFU 3.48 and DFU 3.49 without _any_ issues which is why its interesting that you are both having these issues

Gansgod, 2022-02-07:
This is very strange, I don't know what to do

PUF52, 2022-02-07:
Many are interested in overcoming this problem. She is. I changed computers, changed USB cords, renamed, tried to clean the flash - it's useless. Mystery.

PUF52, 2022-02-07:
Well, if you don't know, I'm all the more stumped. Don't leave us, there are many of us.

Eugene-Market, 2022-02-07:
The firmware is being installed, but nothing is clear on the screen

PUF52, 2022-02-07:
> Я протестировал здесь DFU 3.48 и DFU 3.49 без _каких_ -либо проблем, поэтому интересно, что у вас обоих есть эти проблемы.

Ralim, Can you really connect remotely to my computer and really see what's going on? Can you copy some logs? I can simply provide this access, I just understand that without a soldering iron with such an error on hand, it is difficult to do something remotely, but here there will be direct access.

Eugene-Market, 2022-02-07:
> The firmware is being installed, but nothing is clear on the screen

by the way DFU 3.47

Ralim, 2022-02-08:
@PUF52 I cannot remotely connect to a computer for security reasons.

At this point I dont even know whats best to try since it _looks_ like the bootloader is just crashing partway through flashing, and if its not pc or hardware it makes no sense.

@Eugene-Market 
Have you confirmed your device is the GD32 chip, and are you sure you are installing the correct build of the firmware?

PUF52, 2022-02-08:
Photo of my chip
[image]
I put the firmware that you sent me here

PUF52, 2022-02-08:
дополнительно
[image]
[image]
[image]
[image]

Eugene-Market, 2022-02-08:
> @Eugene-Market Have you confirmed your device is the GD32 chip, and are you sure you are installing the correct build of the firmware?
What is correct firmware? I install 2.16 ts100, I tried languages: Russian, French, English, the screen is empty
on the chip is written the same as in this picture

[image]

PUF52, 2022-02-08:
Ralm, by the way, what version of Windows are you using when testing? Its rank?

PUF52, 2022-02-08:
Bit depth of the system? 32 or 64

JugglerLKR, 2022-02-08:
> @PUF52 I cannot remotely connect to a computer for security reasons.
I can connect to @PUF52 PC, if there is way to get any logs at all. Or list of what to try.

@PUF52 I don't know if it can help, try to put TS100_EN.hex to the root of C:\ drive before flashing. Maybe cyrillic chars interfere somehow?
Try the same with blank.hex first maybe?

JugglerLKR, 2022-02-08:
> @PUF52 I cannot remotely connect to a computer for security reasons.
I can connect to PUF52 PC if there is a way to get any logs somehow. Or what to debug. Is it possible to connect to TS100 DFU with regular usb cable ?

paulfertser, 2022-02-08:
On Tue, Feb 08, 2022 at 01:22:13AM -0800, JugglerLKR wrote:
>   I can connect to PUF52 PC if there is a way to get any logs somehow. Or what
>   to debug. Is it possible to connect to TS100 DFU with regular usb cable ?

Yes, you can take a regular USB cable, cut it and attach to an SWD
debugger, see
https://github.com/Ralim/IronOS/blob/master/Documentation/Flashing.md
"On v2.51A PCB revision USB_D+ is shorted to SWDIO and USB_D- is
shorted to SWCLK so debugging works without disassembly (attach while
staying in the bootloader)".

JugglerLKR, 2022-02-08:
> On Tue, Feb 08, 2022 at 01:22:13AM -0800, JugglerLKR wrote: I can connect to PUF52 PC if there is a way to get any logs somehow. Or what to debug. Is it possible to connect to TS100 DFU with regular [...]

Thank you. I've seen this, but it requires SW debugger hardware which we don't own.

paulfertser, 2022-02-08:
On Tue, Feb 08, 2022 at 01:28:36AM -0800, JugglerLKR wrote:
>   Yes, you can take a regular USB cable, cut it and attach to an SWD debugger,
>   see [1]https://github.com/Ralim/IronOS/blob/master/Documentation/Flashing.md
>   "On v2.51A PCB revision USB_D+ is shorted to SWDIO and USB_D- is shorted to
>   SWCLK so debugging works without disassembly (attach while staying in the
>   bootloader)".
> 
> Thank you. I've seen this, but it requires SW debugger hardware which we don't
> own.

Just about any SBC (including raspberrypi) can bitbang SWD.

PUF52, 2022-02-08:
> Does not help

Ralim, 2022-02-08:
> Ralm, by the way, what version of Windows are you using when testing? Its rank?

I test on Windows 10 x64, Arch Linux and Manjaro Linux as my main "test" setups.
For testing the new bootloader I verified everything on Windows as that is what I assume most people are running.

Honestly the most likely next steps are to use a debugger to read back out what is going on.

Short of that, there is another option, though it will take a bit of work if you are on windows.

So I've been working on [IronOS-dfu](https://github.com/Ralim/IronOS-dfu) after some poking from @discip.

If you are willing to download `dfu-util` and use the command line, you could try using the RUNTIME (not bootloader) version of IronOS-dfu to (a) test flashing the .hex file and (b) take a backup of your bootloader.

Then you could email me the backup of your bootloader and I can load it here and try and reproduce your issue.

If you are willing to do this; you will need to download dfu-util and then follow these steps https://github.com/Ralim/IronOS-dfu/blob/mainline/docs/BackUp.md

PUF52, 2022-02-08:
Guys, sorry for my illiteracy, but the thought that came to my mind is the problem is simpler - how to install firmware 2.16 from ralim on dfu3.48 with a gd32 chip? And any advice about the dfy-util debugger - it.d. they don't say anything to me.

[67 more replies]