Ralim/IronOS #1228: boot logos converted to dfu

River-b, 2022-02-12:
I  got all the Boot logo Art I could find on this github that people shared publicly to be used in pinecil and I converted them to DFU. 
I made a fork with them in the boot logo folder.

How do I upload it or join it into a pull request for review.  It would be nice to include all the Art in one central location like the Boot Logo Folder so people can browse and select what they like to personalize their Pinecil.

Johnex, 2022-02-13:
So you want to fork this repository using the links at the top of the page. You will have a copy in your profile, go there, and click "Add files" in the correct folder. It will let you upload all the files and write what changes you made as a comment. Then from the main page of your copy, you will see a button called "Contribute", and click "Open pull request".

discip, 2022-02-15:
@River-b
You can close this PR, since there is a new approach by the developer on where logos are located and how they are handled.
Due to the fact that it is a WIP, there is no official information on that yet.

Johnex, 2022-02-15:
@River-b You can either delete the file from your repository file list and create a new pull request (delete the old one) or you can delete the repository and try again.

But maybe wait for @Ralim to reply and see where the file should be placed.

PR #1232 (2022-03-13): Logo rework

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [ ] There are no breaking changes

* **What kind of change does this PR introduce?**

Support for new logo support that (a) fixes the old endian issue and (b) allows animation.
Removes logo's from this repo and instead places them in the Meta repo.

* **What is the current behavior?**

* **What is the new behavior (if this is a feature change)?**

* **Other information**:

Closes #1173
Closes #1228
Closes #617
Closes #446
Inches closer to supporting #292
closed 2022-02-15 by River-b