Ralim/IronOS #1229: [TS80P] Sleep mode activates, although iron is in active use

GrigoriOH, 2022-02-13:
**Describe the bug**

While in active use, the sleep timer counts down till sleep activates automatically. Excessive iron motion or display orientation changes do not stop timer. In the debug menu, the "motion" value stays at 0 all the time. Behaviour is not changeable via sensitivity value (tried 7,8,9,1).

**To Reproduce**
1. Activate detailed soldering mode (not needed, just for visualization)
2. Start soldering
3. Wiggle your iron

**Expected behavior**

Sleep times stops when iron is in use / moved.

**Details of your device:**
 
 - Device: TS80P
 - Release: v2.17.E065BE3
 - Power adapter being used: different 5-12V variants
 - Accelerometer: SC7A20 (according to software)

**Additional context**
Motion value was changing in 2.16, although the accelerometer was detected as LIS2DH12 back then (see: #1021)

Ralim, 2022-02-15:
So #1021 fixed the correct ID for this device. Looks like they have hacked up their register map the clone the LIS as well _sigh_.
Ill get a test firmware soonish (this week?).

xia0, 2022-03-10:
Same behaviour here with 2.17 release.
Motion wake works fine on v2.16.72E0C17 but same screen rotation issues and incorrect accelerometer as previously noted.

Ralim, 2022-03-10:
Hia,
Could you give the builds on this action a test? https://github.com/Ralim/IronOS/actions/runs/1962833388

xia0, 2022-03-10:
> Hia, Could you give the builds on this action a test? https://github.com/Ralim/IronOS/actions/runs/1962833388

Just gave it a quick run and appears to work well. Screen orientation works as expected. Sleep timer resets as expected. Motion sensitivity is improved compared to 2.16.

xia0, 2022-03-11:
Upon further testing, I've found that the accelerometer is very sensitive. Even at sensitivity setting of 1, the sleep timer will continually reset even when stationary.

bkeinert, 2022-03-11:
Seems to work perfect for me (TS80P variant with SC7A20 - affected by this issue). Sensitivity setting of 9 - confirmed correct behavior using the debug menu.
I had a quick look over the PR, I think the changes are fine, but potentially a little more than necessary. Wouldn't it be enough to change ```if (FRToSI2C::Mem_Read(SC7A20_ADDRESS, SC7A20_OUT_X_L, (uint8_t *)sensorData, 6) == false) {``` to ```if (FRToSI2C::Mem_Read(activeAddress, SC7A20_OUT_X_L, (uint8_t *)sensorData, 6) == false) {``` in SC7A20.cpp:L74?

Furthermore (not really related to this issue but more to general sensitivity issues), I have another TS80P with MSA301 and noticed the motion detection to be relatively unresponsive (Sensitivity 9). I removed the shifts (/4) on x/y/z in ```MSA301::getAxisReadings()``` (MSA301.cpp:L43-45) leading to much better results. Didn't have a look at the actual x,y,z values from the accelerometers yet. I have the build tool-chain set-up and can experiment with both accelerometer variants.

Ralim, 2022-03-13:
@bkeinert 
>I had a quick look over the PR, I think the changes are fine, but potentially a little more than necessary. 

Correct, but I have a suspicion I'll also need to change the register address we read the accel from, but wasn't sure; so going for toggle felt more flexible.

In terms of the MSA301, happy to take tuning improvements, I do not have one to hand so open to feedback.

@xia0 

So I've pushed the change I held back, that flips the register to read the values from as well.
Can you try the build from here and compare? : https://github.com/Ralim/IronOS/actions/runs/1975574972

bkeinert, 2022-03-13:
> 
> In terms of the MSA301, happy to take tuning improvements, I do not have one to hand so open to feedback.
> 
I'll have a look at actual accelerometer values if I get to it.
 
> @xia0
> 
> So I've pushed the change I held back, that flips the register to read the values from as well. Can you try the build from here and compare? : https://github.com/Ralim/IronOS/actions/runs/1975574972

This change causes "Move" (Debug menu) to just count up on my SC7A20 TS80P (Tested sensitivity 1 & 9). In my experiments a similar combination seemed to work (to some extend) when also sending the "i2c_registers" from LIS2DH12 in initialize() (basically just implementing complete LIS2DH12 behavior, which is known to work to some extend with 2.16 and the SC7A20 variants). That kind of works, but it didn't feel to be responsive enough on Sensitivity 9 for me.

xia0, 2022-03-27:
@Ralim sorry I was away for work for a couple weeks and the build you linked is now expired.

Ralim, 2022-03-27:
Try: https://github.com/Ralim/IronOS/actions/runs/2011640738

xia0, 2022-03-27:
> Try: https://github.com/Ralim/IronOS/actions/runs/2011640738

Just tested now and appears to function correctly. ACC is identified as SC7A20 and sleep timer works normally. Auto screen orientation works normally. Thanks!

bkeinert, 2022-03-27:
> Try: https://github.com/Ralim/IronOS/actions/runs/2011640738

Auto rotation works, sleep timer "kind of works" (as stated before). I really have to move the TS80P harshly to "make it move" and I usually don't use my soldering equipment like a hammer. IMHO not optimal - maybe there are even more sensitivity differences among SC7A20 variants? Not a problem for me, I can build and run whatever works best for me.

Additionally, maybe making sensitivity exponential (settings breaking change) and allowing for a larger range of sensitivities could help. Furthermore, I think it would be good to add a little "*" or something similar to the "ACC SC7A20" in the debug menu to be able to distinguish between the variants - could be useful for future debugging / bug reports.

bkeinert, 2022-03-27:
Update: I played a bit with this and it seems that the SC7A20 really mimics the LIS2DH12 behavior (the current PR is moving in the right direction). I also looked at the actual accelerometer values, it seems that the update rate with `{LIS_CTRL_REG1, 0x17, 0}` is only 1 Hz (I think the 25 Hz value stated in the comment is incorrect, see Table 31 in the LIS2DH12 datasheet). Using `{LIS_CTRL_REG1, 0b00110111, 0}, // 200Hz XYZ` and `{LIS_CTRL_REG2, 0b00000000, 0}` seems get me reasonable accelerometer values on the "fake LIS SC7A20". Looks like it needs the fake LIS setup.

Ralim, 2022-04-02:
Apologies for the delays, @bkeinert could you test https://github.com/Ralim/IronOS/actions/runs/2080627144
I used your tuning values; as I dont have one to test :)

xia0, 2022-04-02:
Accelerometer sensitivity is working fine for me still.

bkeinert, 2022-04-02:
Seems to work just fine for me! I'll build the branch and have a quick look at the actual values from the accelerometer later; just to double check.

closed 2022-04-03 by Ralim

ceut2k4, 2022-12-29:
Hello,
Sorry to post again in this topic.

=>I have bought 3 brand new TS80P at work (from Eleshop), and after putting the 2.20 release,  have seen that none of the 3 goes to standby mode on their stands.
So I checked the Accelerometer, and it is this one too: the "SC7A20" sensor IC.

Mine (an old one I have bought one year ago, with MSA301 IC) has no problem at all.

I have tried the 2.18 version with the "Fixes for SC7A20 in TS80(P)", and still have the same bug.
Even if I set at "1" the Motion sensitivity: the timer in the detailed solder screen doesn't decrease on stand.

So is there something I do wrong ? Or a setting I have not seen ?
(I'm a long time user of your great firmware, with my TS100 I have at home since many years now).

Thank you for your help !

PR #1251 (2022-04-03): SC7A20 tweaks

* **Please check if the PR fulfills these requirements**
- [] The changes have been tested locally
- [] There are no breaking changes

* **What kind of change does this PR introduce?**

Closes #1229 Hopefully.