Ralim/IronOS #1238: MHP30 unable to access sleep settings

weikinhuang, 2022-02-19:
**Describe the bug**
When attempting to go into the sleep settings menu from the main menu, it instead loops showing the sleep settings option of the main menu.

**To Reproduce**
1. Go to settings
2. Click on Sleep Settings
3. Try to enter into sleep settings menu
4. Loops showing sleep settings option from main menu

**Expected behavior**

**Details of your device:**
 - Device: MHP30
 - Release: 2.17
 - Power adapter being used: Anker 60W

Ralim, 2022-02-19:
Ah heck, this is because there are no sleep settings at the moment I believe

weikinhuang, 2022-02-19:
weird, there were def sleep settings in 2.16 though

Ralim, 2022-06-07:
Hia,
So the MHP30 doesnt have sleep mode; but it did have tilt exiting.
I've bought that back (oops) and also added shutdown timer support.
Can you give https://github.com/Ralim/IronOS/actions/runs/2454469240 a test

closed 2022-07-03 by Ralim